espruino/Espruino #1379: Is having crypto build in a waste of space?

MaBecker, 2018-04-24:
For ESP8266 boards only one cypto function  is need if using module WebSockets, which is sha1.

It is a cost of 14kB having crypto build with sha1, sha256 and sha512

```
// with  
** user1.bin uses  464948 bytes of 479232 available
//without
** user1.bin uses  449620 bytes of 479232 available
```
options:
* sha1 could be implemented as Build-in-JS module 
* have sha1 as tinyCrypro lib writen in c

Any ideas or comments?

MaBecker, 2018-04-24:
sample for[ sha1.c](https://github.com/dottedmag/libsha1/blob/master/sha1.c)

opichals, 2018-04-24:
Perhaps we could modularize the `crypto` a bit more with few more #ifdefs like USE_MD_SHA1, USE_MD_SHA256, USE_MD_SHA512 and let it compile-in only the absolutely necessary algorithms.

MaBecker, 2018-04-24:
This would be the smartes way, I guess

wilberforce, 2018-04-24:
Which build are you concerned about? The 512 or 4096 build?

If it's the latter, then getting a build over 512 k that works would be time better spent. The ld file from lua @MaBecker referenced recently might be a good starting point, and I have a feeling the sdk 2.1 might be necessary.

MaBecker, 2018-04-24:
> Which build are you concerned about? The 512 or 4096 build?

just about the fact that using one function means to waste 14KB 

> I have a feeling the sdk 2.1 might be necessary

SDK 2.1 is working fine, no changes to ld files just a few for targets/esp8266/espmissingincludes.h

wilberforce, 2018-04-25:
@mabecker

I thought there was issue that If the firmware is larger than 512k it won’t run?

@opichals

The link above for sha1.js has 2 dependancies, so will be bigger than 2k

Here is code that could be made into a module, and then compiled in using the new js source feature. Might save a bit of space

http://forum.espruino.com/conversations/276175/?offset=25#comment12884444

https://espruino.microco.sm/api/v1/files/fea4a3f2af35c79f21056ec4cf98913619aec40f.js

Not sure this will save much space compare to just including the one function- It’s 1789 bytes

MaBecker, 2018-04-25:
@wilberforce  thanks for sharing sha1.js will try to use this as Build-in-JS module

It is just a question of time till the firmware will be larger than 512k

gfwilliams, 2018-04-25:
I'd go for an ifdef for SHA256/etc. IIRC 192/256 use the same function, and 384/512 use another - so only 2 ifdefs would be needed. It'd be useful for the original Espruino board too.

But yeah, if the 4MB build works it'd be much better to fully load it with all the stuff we can - it's only the 512 that'd be an issue.

MaBecker, 2018-04-25:
Thanks, so lets try something like `USE_ONLY_SHA1=1`   if only SHA1 is wanted.

	modified:   Makefile
	modified:   boards/ESP8266_4MB.py
	modified:   boards/ESP8266_BOARD.py
	modified:   libs/crypto/jswrap_crypto.c
	modified:   libs/crypto/mbedtls/config.h
	modified:   make/targets/ESP8266.make

first build with USE_ONLY_SHA1=1 

	** user1.bin uses  452900 bytes of 479232 available     reduction of 9.152 byte 

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v97.3800 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 512KB:256/256, manuf 0xe0 chip 0x4016
WARNING: *** Your flash chip does not match your flash map ***
>
 {
  "VERSION": "1v97.3800",
  "GIT_COMMIT": "7f678b69",
  "BOARD": "ESP8266_BOARD",
  "FLASH": 0, "RAM": 81920,
  "SERIAL": "5ccf7f05-c100",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,net,dgram,http,NetworkJS,Wifi,ESP8266,TelnetServer,crypto,neopixel",
  "EXPTR": 1073643636 }
 {
  "sdkVersion": "2.0.0(656edbf)",
  "cpuFrequency": 160, "freeHeap": 11000, "maxCon": 10,
  "flashMap": "512KB:256/256",
  "flashKB": 512,
  "flashChip": "0xe0 0x4016"
 }
 { "free": 1600, "usage": 100, "total": 1700, "history": 0,
  "gc": 0, "gctime": 1.419 }
[
  "Flash",
  "Storage",
  "net",
  "dgram",
  "http",
  "NetworkJS",
  "Wifi",
  "ESP8266",
  "TelnetServer",
  "crypto",
  "neopixel"
 ]
=undefined
> 
```

MaBecker, 2018-04-25:
[first commit ](https://github.com/MaBecker/Espruino/commit/0cac4115eb358ebec126ca474100878f0d59bfe0) - what do you think?

Looks like the script build the  jswrapper file  does not care about ifdefs ...

https://github.com/MaBecker/Espruino/blob/0cac4115eb358ebec126ca474100878f0d59bfe0/libs/crypto/jswrap_crypto.c#L168-L246

```
CC gen/jswrapper.o
gen/jswrapper.c:465:15: error: redefinition of 'gen_jswrap_crypto_SHA1'
 static JsVar* gen_jswrap_crypto_SHA1(JsVar* message) {
               ^
gen/jswrapper.c:461:15: note: previous definition of 'gen_jswrap_crypto_SHA1' was here
 static JsVar* gen_jswrap_crypto_SHA1(JsVar* message) {
               ^
```

Had to remove the second JSON SHA1 description part after #else to get around this error

MaBecker, 2018-04-26:
Wow, seem like this was already in your mind
Thanks