espruino/Espruino #1380: rewrite wifi.save and restore to use the store lib

MaBecker, 2018-04-25:
at the moment wifi.save is using a 4kB page to store some credentials and settings.

With the storage library it should be possible to store and restore this data and keep the wifi reconnect at boot time.

@gfwilliams  What do you think, is this  possible?

flash usage now for 512KB boards without FOTA

```
0x7B000		one wifi save page

0x7AFFF
...
0x78000		save code address, 3 pages

0x77000		one flash  area page

0x76FFF
...
0x01000		espruino_esp8266_user1.bin, max 117 pages  = max size 468 KB
```

after moving wifi connection data to storage

```
0x7BFFF	
...
0x78000		save code address, 4 pages

0x77FFF
...
0x01000		espruino_esp8266_user1.bin, max 118 pages  = max size 472 KB
```

gfwilliams, 2018-04-25:
Yes, I'd say that was a great idea - and very possible. Potentially it could be saved as JSON such that it could be parsed and then fed right into the `wifi.connect` function - it'd probably really tidy up the code and maybe even save some code space as well.

wilberforce, 2018-04-25:
If we are going to this trouble, may I suggest a further enhancement?
Store the list of pairs in an array, so that credentials for more than one ssid can be stored.
Potentially we could store the last connect to ssid, and attempt to connect that on restart.

Do we have encryption/decryption - if so we could stored encrypted password rather than plain text password in flash.

MaBecker, 2018-04-26:
Good Point @wilberforce 

Like to start with the config structure and replace  jshFlashWrite and  jshFlashRead with jswrap_storage_read and jswrap_storage_write

Are there some samples how to handle c structures  with storage functions?

gfwilliams, 2018-04-26:
There's no easy encrypt/decrypt we could rely on - after all, if you can read flash you could just read the decryption key as well.

I'd skip trying to save the config structure as-is - looking at the code it's full of hard-coded lengths and things, with strcpys all over the place.

Instead, just do something like this to save:

```
#define WIFI_CONFIG_STORAGE_NAME "wificonf"

JsVar *o = jsvNewObject();
if (!o) return;
jsvSetChildAndUnLock(o, "ssid", jsvNewFromString((char *)sta_config.ssid));
jsvSetChildAndUnLock(o, "password", jsvNewFromString((char *)sta_config.password));
// ...
JsVar *name = jsvNewFromString(WIFI_CONFIG_STORAGE_NAME);
JsVar *arr = jsvNewArray(&o,1);
jswrap_storage_write(name,o,0,0);
jsvUnLock3(arr,name,o);
```

And to load:

```
JsVar *name = jsvNewFromString(WIFI_CONFIG_STORAGE_NAME);
JsVar *arr = jswrap_storage_readJSON(name);
jsvUnLock(name);
if (!arr) return; // no data
JsVar *o = jsvArrayPopFirst(arr); // first array item
jsvUnLock(arr); 
if (!o) return; // no data
JsVar *v;
v = jsvObjectGetChild(o,"ssid",0);
jsvGetString(v, (char *)sta_config.ssid, 32); // it'd be nice not to hard-code the lengths
jsvUnLock(v); 
v = jsvObjectGetChild(o,"password",0);
jsvGetString(v, (char *)sta_config.password, 64); // it'd be nice not to hard-code the lengths
jsvUnLock(v); 
// ...
jsvUnLock(o); 
```

wilberforce, 2018-04-26:
That looks great @gfwilliams 
It looks like the create code will make a new array each time and replace rather than adding on if it is a new array?

This can be optimised later - as long as the initial storage is set up as an array, it can be improved on later.

Should we use pwd instead of password to save a jsvar?

And should the config name be .wificfg so it gets hidden - or should it be visible?

gfwilliams, 2018-04-26:
Yes - to add on you'd have to do a readJSON again and then jsvArrayPush. There's also the potential to iterate over the array when loading and find one that matches available access points, but maybe for the moment I think leave it as-is?

I wouldn't worry about the extra var since it's only momentarily used. As much as possible I'd like to match what http://www.espruino.com/Reference#l_Wifi_connect uses though, just to make things a bit more coherent.

Anyone else got thoughts about `wificonf/.wificfg`? I'm not too bothered - I guess it's vaguely internal so could be hidden?

opichals, 2018-04-26:
I'd vote for the `.wificfg`. Simple rule I tend to apply would be that anything not application specific (Espruino firmware built-in) is hidden.

MaBecker, 2018-04-26:
prefer  ` .wificfg` too

gfwilliams, 2018-04-26:
Ok, cool - maybe put the #define in `jsstorage.h` so that it can be used for every board that stores WiFi info.

opichals, 2018-04-27:
> There's no easy encrypt/decrypt we could rely on - after all, if you can read flash you could just read the decryption key as well.

Could we perhaps at least try to do something as simple as [TEA](https://github.com/espruino/Espruino/issues/1055) just to not to have plaintext? There are going to be boards with security chips to hold the keys in the future.

wilberforce, 2018-04-27:
@opichals 

My thinking was - as it will be a json string - that a simple dump of a rom would not show plain text..

perhaps we are over engineering here?

opichals, 2018-04-27:
> perhaps we are over engineering here?

I wouldn't say so. Even HTTP-Basic style is better than plaintext, right? :)

wilberforce, 2018-04-27:
ok - what about base64 for the keys?

gfwilliams, 2018-04-27:
>  Even HTTP-Basic style is better than plaintext, right? :)

It's just security through obscurity. But it's not even obscure because the whole thing is open source. If someone's going to the point of dumping the flash to look for the keys then it'll take them maybe 30 mins extra to decode - probably no time at all if someone makes a YouTube video or a tool to do it.

All you're doing is making everyone else's life difficult... One of the nice things of having the Wifi info in a user-accessible format is that people can easily write code that accesses it (eg for WiFi setup) without having to actually connect to the WiFi network first.

Possibly even worse than this, by not storing plain text you're fooling people into thinking it's safer than it is. It's much better to just have a warning in `Wifi.save` and to say "do your own thing if you want this secure" - then whoever it is can implement their own complicated by totally insecure system for storing WiFi keys, and it's not our problem.

opichals, 2018-04-27:
@gfwilliams I understand the fact that it is not secure.

It's just a privacy concern, showing the contents of the file if it is base64 at least is somehow _eye safe_. Same as when using a password manager for generating long enough obscure passwords it is mostly not a big deal to actually expose the password (for a brief period) to somebody because there is just no way for a human to remember it.

wilberforce, 2018-04-29:
Looks good.

I'm wondering if the AP and sta modes should be a key stage the root, Rather than combining into all. This way there would be an array for a AP, and one for sta.

Why I does the AP ssid len need to be stored?

MaBecker, 2018-05-05:
second part: [jswrap_wifi_restore](https://gist.github.com/MaBecker/e6600ebd6e647dc4edf32de06b5ca86b)

MaBecker, 2018-05-06:
working:

* wifi.save() and wifi.save("clear") 
* manual wifi.restore()

not working:

* wifi.restore() during reboot

added some jsWarn statements

````
void jswrap_wifi_restore(void) {
  DBG("Wifi.restore\n");

  jsWarn("NAME: %s\n", WIFI_CONFIG_STORAGE_NAME);

  JsVar *name = jsvNewFromString(WIFI_CONFIG_STORAGE_NAME);
  JsVar *o = jswrap_storage_readJSON(name);

  if (!o) { // no data 
    jsvUnLock2(name,o);
    jsWarn("Wifi.restore: NO Data" );
    return; 
  }
.....
````

output during boot
````
jump to run user1 @ 1000
âìnìròn|ìll`ârlþWARNING: NAME: .wificfg
WARNING: Wifi.restore: NO Data
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v97.3800 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:1024/1024, manuf 0xe0 chip 0x4016
> 
````

Could need some help to get this working....

wilberforce, 2018-05-06:
What order are you doing this in? Has everything's been initialized that needs to before you read from storage?
Does it work if you do it last after a Delay?

MaBecker, 2018-05-07:
> Has everything's been initialized that needs to before you read from storage?

Good question, I would say  NO.

> What order are you doing this in?

Order did not change, called from [user_main.c](https://github.com/espruino/Espruino/blob/master/targets/esp8266/user_main.c#L343).

> Does it work if you do it last after a Delay?

Added   os_delay_us(1000) and move   jswrap_ESP8266_wifi_init1() to the end of function user_init,
still NO data

wilberforce, 2018-05-07:
try 5 secs.....

MaBecker, 2018-05-07:
@gfwilliams any hints from your side?

gfwilliams, 2018-05-07:
If you're doing this from `user_init` it's probably before the JsVars have initialised, so trying to create a string will probably just return 'undefined',

Maybe try it from here? https://github.com/espruino/Espruino/blob/master/libs/network/esp8266/jswrap_esp8266_network.c#L1205