espruino/Espruino #1391: wifi reconnecting  permanently - Is this new or does it work as designed?

MaBecker, 2018-05-02:
During tests for the new wifi.save() and wifi.restore() this issues was detected.

* One ESP is setup as AP and the other as STA

* STA ESP disconnects and reconnects 

No idea if this was the case since beginning or during code updates. 

Any hints why this disconnect and reconnects happens nearly ever second?

My understanding is that the connection station to access point is connected and than stable without any disconnects.

TestCode:

```
require('ESP8266').logDebug(true);
require('ESP8266').setLog(2);

STA = "5ccf7fc1-120b";
AP =  "5ccf7f05-c100";

var Wifi = require('Wifi');

SSID='ESPAP';
SSIDPASS='espwlan.';

console.log("My serial is:", getSerial());
console.log('\n',process.env,'\n',
            require("ESP8266").getState(),
            '\n',process.memory());

console.log(process.env.MODULES.split(','));

if ( getSerial() === STA ) {
  console.log("I am a Station");
  Wifi.connect(SSID, {password:SSIDPASS}, function(err){
    if (err === null){
      console.log('Connected to AP "'+SSID+'"');
      Wifi.stopAP();
      Wifi.setConfig({powersave : 'none'});
    }
    else {
      console.log('Got error:',err);
      require('Wifi').disconnect();
    }
  });
}

if ( getSerial() === AP ) {
  console.log("I am a Access Point ");
  Wifi.setConfig({powersave : 'none'});
  Wifi.startAP(SSID, {password:SSIDPASS,channel:2}, function(err) {
    if (err === null){
      console.log('Start AP "'+SSID+'"');
    } else {
      console.log('Got error:',err);
    }
  });
}
```

Log on  station:
  
```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v97.3812 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:1024/1024, manuf 0xe0 chip 0x4016
>My serial is: 5ccf7fc1-120b
>
 {
  "VERSION": "1v97.3812",
  "GIT_COMMIT": "2861d7ef",
  "BOARD": "ESP8266_4MB",
  "FLASH": 0, "RAM": 81920,
  "SERIAL": "5ccf7fc1-120b",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,net,dgram,http,NetworkJS,Wifi,ESP8266,TelnetServer,crypto,neopixel",
  "EXPTR": 1073643636 }
 {
  "sdkVersion": "2.0.0(656edbf)",
  "cpuFrequency": 160, "freeHeap": 12344, "maxCon": 10,
  "flashMap": "4MB:1024/1024",
  "flashKB": 4096,
  "flashChip": "0xe0 0x4016"
 }
 { "free": 1484, "usage": 116, "total": 1600, "history": 30,
  "gc": 0, "gctime": 1.814 }
[
  "Flash",
  "Storage",
  "net",
  "dgram",
  "http",
  "NetworkJS",
  "Wifi",
  "ESP8266",
  "TelnetServer",
  "crypto",
  "neopixel"
 ]
I am a Station
 27584> mode : sta(5c:cf:7f:c1:12:0b) + softAP(5e:cf:7f:c1:12:0b)
 27585> add if0
=undefined
> 29804> scandone
 30687> state: 0 -> 2 (b0)
 30689> state: 2 -> 3 (0)
 30692> state: 3 -> 5 (10)
 30692> add 0
 30692> aid 1
 30692> cnt
 30702>
 30702> connected with ESPAP, channel 2
 30703> dhcp client start...
 31887> ip:192.168.4.2,mask:255.255.255.0,gw:192.168.4.1
Connected to AP "ESPAP"
 31890> bcn 0
 31890> del if1
 31890> pm open,type:2 0
 31890> mode : sta(5c:cf:7f:c1:12:0b)
 31988> sleep disable
> 52689> state: 5 -> 2 (4a0)
 52689> rm 0
 52689> pm close 7
 53689> reconnect
 53689> state: 2 -> 0 (0)
 53810> scandone
 53810> state: 0 -> 2 (b0)
 53812> state: 2 -> 3 (0)
 53817> state: 3 -> 5 (10)
 53817> add 0
 53817> aid 1
 53817> cnt
 53827>
 53827> connected with ESPAP, channel 2
 53827> dhcp client start...
 54713> ip:192.168.4.2,mask:255.255.255.0,gw:192.168.4.1
 63818> pm open,type:0 0
....
```

Log accespoint: 

```

 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v97.3800 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 512KB:256/256, manuf 0xe0 chip 0x4016
WARNING: *** Your flash chip does not match your flash map ***
>My serial is: 5ccf7f05-c100
>
 {
  "VERSION": "1v97.3800",
  "GIT_COMMIT": "7f678b69",
  "BOARD": "ESP8266_BOARD",
  "FLASH": 0, "RAM": 81920,
  "SERIAL": "5ccf7f05-c100",
  "CONSOLE": "Serial1",
  "MODULES": "Flash,Storage,net,dgram,http,NetworkJS,Wifi,ESP8266,TelnetServer,crypto,neopixel",
  "EXPTR": 1073643636 }
 {
  "sdkVersion": "2.0.0(656edbf)",
  "cpuFrequency": 160, "freeHeap": 10024, "maxCon": 10,
  "flashMap": "512KB:256/256",
  "flashKB": 512,
  "flashChip": "0xe0 0x4016"
 }
 { "free": 1584, "usage": 116, "total": 1700, "history": 30,
  "gc": 0, "gctime": 1.462 }
221446> add 1
221447> aid 1
221447> station: 5c:cf:7f:c1:12:0b join, AID = 1
[
  "Flash",
  "Storage",
  "net",
  "dgram",
  "http",
  "NetworkJS",
  "Wifi",
  "ESP8266",
  "TelnetServer",
  "crypto",
  "neopixel"
 ]
I am a Access Point
224014> sleep disable
=undefined
Start AP "ESPAP"
>241446> station: 5c:cf:7f:c1:12:0b leave, AID = 1
241446> rm 1
242570> add 1
242570> aid 1
242570> station: 5c:cf:7f:c1:12:0b join, AID = 1
262570> station: 5c:cf:7f:c1:12:0b leave, AID = 1
262570> rm 1
263695> add 1
263695> aid 1
263695> station: 5c:cf:7f:c1:12:0b join, AID = 1
.....
```

wilberforce, 2018-05-02:
Why don't you flash the boards with an earlier firmware before your changes and see if the condition still occurs?

MaBecker, 2018-05-02:
Thanks @wilberforce yes did this too.

Finding: It is the combination of ESP-STA connect to ESP-AP.

Created a AP on a cell phone and had no more automatic reconnects with an ESP-STA.

closed 2018-05-02 by MaBecker