espruino/Espruino #1402: Add GRAPHICS to ESP8266_BOARD

MaBecker, 2018-05-08:
With #1380 it is possible to add GRAPHICS, but every next change could over step the max build size.

Only 524 byte left :-(

````
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  3 .text         00007b88  40100000  40100000  00002d44  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   0006b5c4  40201010  40201010  0000a8d0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf >espruino_esp8266_user1.lst
  8 -rwxr-xr-x  1 markbecker  staff    2188  8 Mai 08:09 eagle.app.v6.data.bin
864 -rwxr-xr-x  1 markbecker  staff  439748  8 Mai 08:09 eagle.app.v6.irom0text.bin
 24 -rwxr-xr-x  1 markbecker  staff    9172  8 Mai 08:09 eagle.app.v6.rodata.bin
 64 -rwxr-xr-x  1 markbecker  staff   31624  8 Mai 08:09 eagle.app.v6.text.bin

** user1.bin uses  482804 bytes of 483328 available

````

Question: 

* This there a way to split  GRAPHICS  like done for CRYPTO?

* Or are the other section that can be optimized to reduce space?

gfwilliams, 2018-05-09:
Yes, there probably is a way, however if SAVE_ON_FLASH is enabled then Graphics will automatically do some things to help with this like removing vector fonts.

I don't want to add Graphics to a build that's this tight though because as you say any subsequent change could push it over - and end up being it'll be me that has to fix it :)

Can you not use the 4MB build, or is that broken? Even the ESP01 actually has 1MB of flash that could be used - you just wouldn't get the OTA updates but few people use those anyway.

MaBecker, 2018-05-09:
> Yes, there probably is a way, however if SAVE_ON_FLASH is enabled then Graphics will automatically do some things to help with this like removing vector fonts.

Did some tests:

    BOARD=ESP8266_BOARD 
    ** user1.bin uses  453092 bytes of 479232 available
    
    plus SAVE_ON_FLASH=1 
    ** user1.bin uses  431348 bytes of 479232 available    

    plus GRAPHICS
    ** user1.bin uses  445060 bytes of 479232 available

> I don't want to add Graphics to a build that's this tight though because as you say any subsequent change could push it over - and end up being it'll be me that has to fix it :)

 I fully agree, this was also never in my mind, just have been  nosy if it would fit.

> Can you not use the 4MB build, or is that broken?

BOARD_4MB is ok

Was looking for a ESP8285 solution, but as you said all small boards now have at least 1MB

Will run some tests with ESP8266_BOARD and FLASH_1MB