espruino/Espruino #1406: align ESP8266 Wifi calls with ESPRUINOWifi calls

MaBecker, 2018-05-09:
Since this commit eaef9c1440a0d634202b61097fc814921a8e6aae

    Add better docs for the form of Wifi callback functions 
    Modify ESP8266/ESP32 callbacks to match the node.js style used elsewhere

some ESP8266 callback need to be aligned

@wilberforce is ESP32 aligned ?

gfwilliams, 2018-05-09:
Aligned? I think I changed the callbacks when I made that change?

MaBecker, 2018-05-09:
Here is one example for Wifi.scan(...) output should be same 

 Espruino Wifi

````
  {
    "ssid": "Dialog 4G",
    "authMode": "wpa_wpa2_psk",
    "rssi": -72,
    "mac": "7c:11:cb:08:7a:f3",
    "channel": 11 }
````

  ESP8266
````
  { "rssi": -46, "channel": 1, "authMode": 0, "isHidden": false,
    "ssid": "ESP_05C100",
    "mac": "5e:cf:7f:05:c1:00"
   }
````

gfwilliams, 2018-05-09:
Ahh right - well they're not too different I guess. I wouldn't worry about extra items and things - but it'd be great to change stuff like `authmode` and/or change the docs to show things that are specific to certain platforms

wilberforce, 2018-05-09:
I think the authmode on the esp8266 could be converted to a string - I think the look up tables already exists - although the string might not be exact.

wilberforce, 2018-05-09:
>. @wilberforce is ESP32 aligned ?

I've not made any changes if that's what you are asking?

MaBecker, 2018-05-09:
>I think the authmode on the esp8266 could be converted to a string

Yep lookup table is their, will implement this and compare all Wifi outputs.

> I've not made any changes if that's what you are asking?

So ESP32 output is same as ESP8266

MaBecker, 2018-05-10:
And you could add `ishidden` too as false or true like ESP8266, because ESP8266 can now [scan](https://github.com/espruino/Espruino/blob/master/libs/network/esp8266/jswrap_esp8266_network.c#L473-L475) for hidden access points.

MaBecker, 2018-07-03:
@wilberforce anything you like to add?

wilberforce, 2018-07-03:
@MaBecker 

Not sure what you are asking me?  I pointed out differences  in the string in the comments above (d) however not-one commented.

MaBecker, 2018-07-04:
esp8266 part is done. 

@wilberforce close or keep open?

closed 2018-07-04 by wilberforce