espruino/Espruino #1427: stack over error

HongxuChen, 2018-05-19:
We found with our fuzzer a stackoverflow when the input file contains many parentheses.

```
ASAN:DEADLYSIGNAL
=================================================================
==28915==ERROR: AddressSanitizer: stack-overflow on address 0x7fffad975fb0 (pc 0x55fb67713a27 bp 0x7fffad976050 sp 0x7fffad975f30 T0)
    #0 0x55fb67713a26 in jspeFactor src/jsparse.c:1569
    #1 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #2 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #3 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #4 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #5 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #6 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #7 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #8 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #9 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #10 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #11 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #12 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #13 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #14 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #15 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #16 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #17 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #18 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #19 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #20 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #21 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #22 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #23 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #24 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #25 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #26 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #27 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #28 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #29 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #30 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #31 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #32 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #33 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #34 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #35 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #36 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #37 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #38 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #39 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #40 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #41 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #42 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #43 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #44 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #45 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #46 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #47 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #48 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #49 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #50 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #51 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #52 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #53 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #54 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #55 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #56 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #57 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #58 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #59 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #60 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #61 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #62 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #63 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
[... 169 lines cut ...]
    #233 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #234 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #235 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #236 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #237 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #238 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #239 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #240 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #241 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #242 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765
    #243 0x55fb67715ac6 in jspeUnaryExpression src/jsparse.c:1791
    #244 0x55fb6771616a in jspeBinaryExpression src/jsparse.c:1919
    #245 0x55fb677163c5 in jspeConditionalExpression src/jsparse.c:1955
    #246 0x55fb67716b01 in jspeAssignmentExpression src/jsparse.c:2020
    #247 0x55fb67713080 in jspeExpressionOrArrowFunction src/jsparse.c:1485
    #248 0x55fb6771407e in jspeFactor src/jsparse.c:1606
    #249 0x55fb67711548 in jspeFactorFunctionCall src/jsparse.c:1200
    #250 0x55fb6771576f in jspePostfixExpression src/jsparse.c:1765

SUMMARY: AddressSanitizer: stack-overflow src/jsparse.c:1569 in jspeFactor
==28915==ABORTING
```

sample input file:
[so_0.txt](https://github.com/espruino/Espruino/files/2019243/so_0.txt)

HongxuChen, 2018-05-19:
This was triggered still with address sanitizer.

With the input script you provided, the output on my console is:

```
Uncaught SyntaxError: Unexpected end of Input
```

gfwilliams, 2018-05-19:
That's odd - I wonder whether `jsuGetFreeStack` isn't working on your build for some reason? It might explain your errors.

closed 2018-05-21 by gfwilliams

gfwilliams, 2018-05-21:
Thanks - that was a stupid mistake. Must have just been luck that it worked on mine. It looks like your stack frames are bigger with the AddressSanitizer, so the 512 byte headroom check can sometimes never catch that before it overflows.

commit a0d7f432 (2018-05-21): Fix stack size detection on Linux (fix #1427)