espruino/Espruino #1432: Missing require('Storage').readStream('file')

opichals, 2018-05-20:
It'd be nice way to stream data out in WebServer via `pipe()` call

Also this can allow processing larger than 65kB contents (as strings are limited to this).

opichals, 2018-05-24:
> With the 65k limit, I think we can actually get away without that on all but the most confined (<1k vars) platforms. I've just tweaked it.

I don't understand. How can the data beyond the 65k limit /that can be written by require('Storage').write('file', "test", 66000, 100)/ be accessed?

opichals, 2018-05-24:
And yes, the JS module is completely fine for this, not sure what would be more space-saving.

gfwilliams, 2024-10-10:
Closing this as you can now do `E.pipe(require('Storage').read(...), dst)` and this should work fine

closed 2024-10-10 by gfwilliams