espruino/Espruino #1434: Stack overflow error

HongxuChen, 2018-05-21:
We found another stack overflow error on 16f3d99f7e089538a20417bf348e75e3c5c19b15  that is different from #1427. The input file is rather contrived (due to fuzzing) but it can be triggered without address sanitizer.

sample input file:
[input.txt](https://github.com/espruino/Espruino/files/2022588/input.txt)

gdb backtrace:
```
Thread 1 "espruino" received signal SIGSEGV, Segmentation fault.
0x0000555555576f41 in jspeFactor () at src/jsparse.c:1719
1719	    JSP_ASSERT_MATCH(LEX_R_VOID);
#0  0x0000555555576f41 in jspeFactor () at src/jsparse.c:1719
#1  0x00005555555750e4 in jspeFactorFunctionCall () at src/jsparse.c:1200
#2  0x00005555555771cb in jspePostfixExpression () at src/jsparse.c:1765
#3  0x0000555555577382 in jspeUnaryExpression () at src/jsparse.c:1791
#4  0x0000555555576f50 in jspeFactor () at src/jsparse.c:1720
#5  0x00005555555750e4 in jspeFactorFunctionCall () at src/jsparse.c:1200
#6  0x00005555555771cb in jspePostfixExpression () at src/jsparse.c:1765
#7  0x0000555555577382 in jspeUnaryExpression () at src/jsparse.c:1791
#8  0x0000555555576f50 in jspeFactor () at src/jsparse.c:1720
#9  0x00005555555750e4 in jspeFactorFunctionCall () at src/jsparse.c:1200
#10 0x00005555555771cb in jspePostfixExpression () at src/jsparse.c:1765
#11 0x0000555555577382 in jspeUnaryExpression () at src/jsparse.c:1791
#12 0x0000555555576f50 in jspeFactor () at src/jsparse.c:1720
#13 0x00005555555750e4 in jspeFactorFunctionCall () at src/jsparse.c:1200
#14 0x00005555555771cb in jspePostfixExpression () at src/jsparse.c:1765
#15 0x0000555555577382 in jspeUnaryExpression () at src/jsparse.c:1791
#16 0x0000555555576f50 in jspeFactor () at src/jsparse.c:1720
...
#95144 0x0000555555576f50 in jspeFactor () at src/jsparse.c:1720
#95145 0x00005555555750e4 in jspeFactorFunctionCall () at src/jsparse.c:1200
#95146 0x00005555555771cb in jspePostfixExpression () at src/jsparse.c:1765
#95147 0x0000555555577382 in jspeUnaryExpression () at src/jsparse.c:1791
#95148 0x00005555555778da in jspeBinaryExpression () at src/jsparse.c:1919
#95149 0x0000555555577aa1 in jspeConditionalExpression () at src/jsparse.c:1955
#95150 0x0000555555577e6d in jspeAssignmentExpression () at src/jsparse.c:2020
#95151 0x0000555555577e8b in jspeExpression () at src/jsparse.c:2026
#95152 0x000055555557a317 in jspeStatement () at src/jsparse.c:2673
#95153 0x00005555555780d6 in jspeBlockOrStatement () at src/jsparse.c:2079
#95154 0x000055555557814c in jspParse () at src/jsparse.c:2091
#95155 0x000055555557ad3d in jspEvaluateVar (str=0x7ffff7fd4170, scope=0x0, lineNumberOffset=0x0) at src/jsparse.c:2899
#95156 0x000055555557af01 in jspEvaluate (str=0x555555812490 "E>-ArrayBuff", 'e' <repeats 63 times>, "n", 'e' <repeats 124 times>..., stringIsStatic=0x0) at src/jsparse.c:2931
#95157 0x00005555555c58da in main (argc=0x2, argv=0x7fffffffb6e8) at targets/linux/main.c:330
```

gdb backtrace file (full):
[gdb.txt](https://github.com/espruino/Espruino/files/2022587/gdb.txt)

gfwilliams, 2018-05-21:
Thanks - that's one bonus of having https://github.com/espruino/Espruino/issues/1428 - you're much more likely to be able to hit issues with the fuzzing :)

closed 2018-05-21 by gfwilliams

HongxuChen, 2018-05-21:
Yes, exactly.

catch22out, 2020-08-29:
abaaba

commit c36d3052 (2018-05-21): Fix stack overflow if void void void... is repeated many times (fix #1434)