espruino/Espruino #1436: No text rendered in screen (SH1106 i2c) with ESP32 espruino 1v98

RndMnkIII, 2018-05-21:
I've tested the following example running from Web IDE version 0.68.6 Windows 64bit with ESP32 DevKitV1 (4Mb FLASH):
`//require("SH1106");
var graficos;

// I2C
I2C2.setup({scl:D22,sda:D21, bitrate:400000});
graficos = require("SH1106").connect(I2C2, function(){});

console.log("hola");
for(var i=1; i<100; i++)
{
  dibuja(i);
  setTimeout(function(){},100);
}

function dibuja(i){
 graficos.clear();
 graficos.drawString("World",0,10);
 graficos.drawLine(i, 5, 10+i, 60);
 graficos.flip();
}`
The example only draws the line  but not the string. It's the  drawString not working???
[image]

wilberforce, 2018-05-21:
Please post these kind of questions in the ESP32 forum.

Please try the simple case - remove the loop and timeouts and see what you get. I think you a `flip()` ing to fast.

I don't have the hardware, however this redenders to an array bugger and you can see the output:

```
//I2C2.setup({scl:D22,sda:D21, bitrate:400000});
//graficos = require("SH1106").connect(I2C2, function(){});

Graphics.prototype.print = function() {
  for (var y=0;y<this.getHeight();y++)
    console.log(new Uint8Array(this.buffer,this.getWidth()*y,this.getWidth()).toString());
};
var graficos = Graphics.createArrayBuffer(30,30,8);

graficos.setColor(1);
console.log("hola");

function dibuja(i){
graficos.clear();
//graficos.drawString("World",0,10);
//graficos.drawLine(i, 5, 10+i, 60);
//graficos.flip();
//graficos.drawString("Hi",0,0);
graficos.drawString("World",0,10);
graficos.print();
}

dibuja(1);
```

references:

http://forum.espruino.com/conversations/287196/#comment13070259
http://www.espruino.com/Graphics#line=22

closed 2018-05-22 by gfwilliams

gfwilliams, 2018-05-22:
There'll be a 1v99 release with some changes in soon that may well fix this for you (or you can use the travis builds).

But as @wilberforce says please post questions like this in the forum in the future. It makes it far more likely someone will be able to help you, as in 99% of cases these problems aren't related to the Espruino interpreter but might be related to the software module for the OLED or the hardware setup itself.

commit a457bd0c (2018-05-22): Fix font rendering issue caused by signed bit field handling by GCC on non-x86 platforms (fix #1436)