espruino/Espruino #1440: changeInterval dont change immediatly but after the last interval

jbgagnon, 2018-05-23:
function testChangeInterval() {
    var t0 = Date.now();
    var i = 0;
    var id = setInterval(() => {
        var t1 = Date.now();
        console.log('has wait', t1 - t0);
        t0 = t1;
        var zz = (( i++ % 3) + 1) * 5000;
        console.log('need wait', zz);
        changeInterval(id, zz);
    }, 1000);
}

this fonction log:
has wait 1002.05078125
need wait 5000
has wait 999.9521484375
need wait 10000
has wait 5000.03686523437
need wait 15000
has wait 10000.03686523437
need wait 5000
has wait 14999.6611328125
need wait 10000
has wait 5000.2109375
need wait 15000
has wait 10000.06811523437
need wait 5000
has wait 15000.02783203125
and so one

solution: jsiTimersChanged() must recalculate the current time to idle and not wait the next time

the documentation  says :
"This takes effect immediately and resets the timeout, so in the example above, regardless of when you call"

jbgagnon, 2018-05-23:
perhaps a solution :

I am too new to know if is it multi-thread but when I see this code  in jsiIdle

jsiStatus = jsiStatus & ~JSIS_TIMERS_CHANGED;
  JsVar *timerArrayPtr = jsvLock(timerArray);
  JsvObjectIterator it;
  jsvObjectIteratorNew(&it, timerArrayPtr);
  while (jsvObjectIteratorHasValue(&it) && !(jsiStatus & JSIS_TIMERS_CHANGED)) { 

how can jsiStatus have bit JSIS_TIMERS_CHANGED if we remove it first by
jsiStatus = jsiStatus & ~JSIS_TIMERS_CHANGED;
(that can be write jsiStatus &= ~JSIS_TIMERS_CHANGED; )

but if so

in  _jswrap_interface_setTimeoutOrInterval or _jswrap_interface_clearTimeoutOrInterval jsiTimersChanged is called when the timerArray is unlocked

 and in jswrap_interface_changeInterval jsiTimersChanged is called when timerArray is still locked

hope this help

gfwilliams, 2018-05-23:
Espruino isn't multithreaded but JS code will be executed during the loop of `timerArray`, which could call `changeInterval`. I imagine it's not re-reading the `time` and `interval` values. 

If this is causing you a problem I'd switch to using repeated calls to `setTimeout` for now - which would probably be more efficient.