espruino/Espruino #1441: Feature or over-optimization?

HongxuChen, 2018-05-23:
We found an input that will hang espruino:

```javascript
let d = 0;
while (10) {
  if (d < 7) continue;
  wrong here // parse error
}
```

In my opinion the `while` construct should be parsed as a whole and we were expecting espruino to report syntax parsing error, however it interprets this bulk of code as if the following line does not exist.

On the other hand, the code below reports no error:

```javascript
let d = 0;
while (10) {
  if (d < 7) break;
  wrong here
}
```

And this code reports Uncaught SyntaxError: Got ID:devNum expected ';'``.

```javascript
let d = 0;
while ( 10) {
  if (d < 7) break;
  for (var i = 0;i devNum; i++) {
  }
}
```

But I'm not sure about the design philosophy of espruino. So is this designed intentionally or a mistake?

gfwilliams, 2018-05-23:
Interesting, thanks - I'll look into the non-breakableness.

re. lack of parse errors: It's designed that way. Basically Espruino parses as it executes (it's designed for devices with so little RAM there's no room to reliably allocate a parse tree or bytecode). If it never gets to that point it won't execute.

Having said that, in some ways it is under-optimised - it will continue to parse a basic block that it *started* executing, it just won't bother to look up variables or actually do the work.

In fact the `wrong here` code you're executing would be treated as being ok anyway. And *that* is a byproduct of not implementing JS's semicolon insertion and so working around it. eg:

```
{ 1 2 }
```

has to be treated the same as:

```
{ 1 
2 }
```

And if a value isn't used, Espruino doesn't bother creating a ReferenceError for it... Maybe it should though.

closed 2018-05-23 by gfwilliams

HongxuChen, 2018-05-23:
OK, I got it. It postponed the syntax checking by keeping the matching pairs such as `{`.

HongxuChen, 2018-05-23:
Does 9221b65900831a8e4600b85679c6279bde25ced1 check reference error? Seems not when tested on my machine on 2nd example.

gfwilliams, 2018-05-23:
You mean:

```
let d = 0;
while (10) {
  if (d < 7) break;
  wrong here
}
```

It won't, because it'll never actually execute the given code (eg, no symbol table lookups).

HongxuChen, 2018-05-23:
OK, I see.

commit 3b7739af (2018-05-23): Allow Ctrl-C out of while...continue loop (fix #1441)