espruino/Espruino #1444: Util Timer enhancements

gfwilliams, 2018-05-24:
- [ ] Currently we use a 64 bit time in the util times, which really slows things down as we're doing 64 bit maths inside the IRQ. We could move to 32 bits, perhaps switching to a 'time until scheduled' method rather than storing the raw system time.

- [ ] We also use a queue which is shuffled around to reschedule, requiring a reasonably big copy each time. We could keep this constant and just have an array of indexes that we reordered. Another benefit of this would be that timers had IRQs, so could be easily checked in the code.

- [ ] Some things (like waveform) continually poll to see if the waveform is active or not. The waveform could actually just post to the event queue (using the new timer IDs) to avoid this polling.

- [ ] When doing ADC reads, we wait in the IRQ for the read to complete. The HAL could be modified to allow an ADC read with callback function + user data, which would make waveform to take up a lot less time.

wilberforce, 2018-05-24:
Is it worth benchmarking the 32 bit vs 64 bit for point 1 to see what the real gains are? I'm sure this will be different for each platform too.

gfwilliams, 2020-10-07:
Done some work on this - as you note the hit from 64->32 math isn't huge. There hasn't been a massive improvement.

The one that hurt was recurring timers which used what ended up being a 64 bit divide, so it's nice to get rid of that.