espruino/Espruino #1448: Stack over flow error with certain number of "{" in input file

HongxuChen, 2018-05-27:
When there are certain number of continuous "{" inside input file, it may cause some stackoverflows. 

For example, on my machine (Linux x86_64) when compiled with address sanitizer, espruino emits stack overflow when there are 130671, 130676, 130683, 130747, ... >130774 continuous "{" in the input files. 

This was actually observed by our fuzzer in Espruino's older versions several times, but we didn't reproduce with vanilla address sanitizer. We wrote the following simple script to generate the input and finally reproduced it on ca13e4d02c53959d2cfbc82bad7853e5a8f5eb27.

```python
#!/usr/bin/env python

import sys
import subprocess

if len(sys.argv) < 2:
    print("wrong cmd, exit")
    sys.exit(1)
else:
    exe = sys.argv[1]
    print(exe)

def write_file(n, fname):
    with open(fname, "w") as f:
        f.write("{" * n)

def call(fname):
    out_fname = fname + ".res"
    cmd_str = "{} {}".format(exe, fname)
    print(cmd_str)
    cmd = cmd_str.split()
    with open(out_fname, "w") as out_f:
        c = subprocess.call(cmd, stdout=out_f, stderr=out_f)

for i in xrange(130650, 300000):
    fname = "{:08d}".format(i)
    write_file(i, fname)
    call(fname)

```

HongxuChen, 2018-05-27:
BTW, it seems that *whenever* there are some crashes (although only in few scenarios), espruino will ruin the terminal, it there any way to mitigate this?

[image]

MaBecker, 2018-05-27:
>  it there any way to mitigate this?

try  `stty sane`

HongxuChen, 2018-05-27:
@MaBecker Thanks :smile:

commit 51380baf (2018-05-28): Fix stack overflow if interpreting a file full of '{' (fix #1448)
closed 2018-05-28 by gfwilliams