espruino/Espruino #1451: Stack over flow error in jsvCopy

HongxuChen, 2018-05-28:
We found with our fuzzer several segmentation faults at src/jsvar.c:2373 ( `jsvCopy`) on the latest espruino (00b1649b79a7ae605d5e4394ca8d3708edbab575). When compiled with Address sanitizer, it indicates it's a stack over flow error.

```
ASAN:DEADLYSIGNAL
=================================================================
==4052==ERROR: AddressSanitizer: stack-overflow on address 0x7ffc1d02cfd8 (pc 0x7f2d11c626c9 bp 0x7ffc1d02d870 sp 0x7ffc1d02cfe0 T0)
    #0 0x7f2d11c626c8  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0x796c8)
    #1 0x5589aad691f5 in jsvCopy src/jsvar.c:2373
    #2 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #3 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #4 0x5589aad6933c in jsvCopy src/jsvar.c:2399
...
    #244 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #245 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #246 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #247 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #248 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #249 0x5589aad6933c in jsvCopy src/jsvar.c:2399
    #250 0x5589aad6933c in jsvCopy src/jsvar.c:2399

SUMMARY: AddressSanitizer: stack-overflow (/usr/lib/x86_64-linux-gnu/libasan.so.4+0x796c8) 
==4052==ABORTING
```

Sample input files:
[test_01.txt](https://github.com/espruino/Espruino/files/2044791/test_01.txt)
[test_02.txt](https://github.com/espruino/Espruino/files/2044792/test_02.txt)

gfwilliams, 2018-05-28:
Thanks - this one's a bit contrived as it's just filling up all available RAM with string data. In an embedded environment I imagine we'd run out of RAM for variables before the stack complained.

However `jsvCopy` could be modified not to recurse when StringExts are involved, which I think would be an improvement all around.

HongxuChen, 2018-05-28:
Yes it's contrived and the original seed was taken from another project for general purposed javascript interpretation. Maybe you can document this caveat a bit.
I'm not sure whether specialization of StringExts benefits much though.

commit 73480706 (2018-05-31): Switch to non-recursive StringExt copy (fix #1451)
closed 2018-05-31 by gfwilliams