espruino/Espruino #1455: Use of Potentially Dangerous Function - CWE-676

teamwoot, 2018-06-02:
Hi Team, 

We observed that `https://github.com/espruino/Espruino/blob/master/targets/linux/jshardware.c#L302`

i.e

`usleep(shortSleep ? 1000 : 50000);`

This C routine is considered obsolete (as opposed to the shell command by the same name). The interaction of this function with SIGALRM and other timer functions such as sleep(), alarm(), setitimer(), and nanosleep() is unspecified such as (CWE-676). 

Use nanosleep(2) or setitimer(2) instead.

Request team to have a look for same.

Regards
Team w00t

wilberforce, 2018-08-03:
@gfwilliams 

Shall we close this? Are you likely to change `sleep` in the Linux server build?

gfwilliams, 2018-08-03:
It wouldn't be a big change, so I'm tempted to do it. I wonder whether at some point the fact it's obsolete will mean that there start being issues when compiling it for embedded linux targets.

wilberforce, 2018-08-06:
Looks like there is a function you could use to replaced defined in the mbed libs:

https://github.com/espruino/Espruino/search?q=usleep&unscoped_q=usleep

gfwilliams, 2018-08-07:
Thanks - just sorted

commit 85a5fb77 (2018-08-07): Replace use of obsolete 'usleep' function in Linux builds (fix #1455)
closed 2018-08-07 by gfwilliams