espruino/Espruino #1457: Storage write of boot code crashes

sirpy, 2018-06-04:
using v1.99
I'm trying to take a backup of the boot code before updating, I get the following error:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8
2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
âÜì8nÀä>rÜØònãblrâÜçÉ8æRunning onInit()...
```

the code to copy the old boot is:
var oldCode = Storage.read(".bootrst")
Storage.write(".bootbac",oldCode,0);

gfwilliams, 2018-06-04:
So I guess this is ESP8266? Did you read the boot code in the same version you wrote it?

What are you updating? The Espruino firmware, or your code itself?

Does the read itself work? If you do: `print(JSON.stringify(require("Storage").read(".bootrst")))`

sirpy, 2018-06-05:
I'm trying to update just the code... the boot code that is uploaded with the IDE, i'm trying to build an OTA update of my code.
Yes the reading works.
Yes it's ESP8266.
If I type the following code in the WEB IDE prompt then it crashes just the same:
var Storage = require("Storage")
var oldCode = Storage.read(".bootrst")
Storage.write(".bootbac",oldCode,0,oldCode.length);

gfwilliams, 2018-06-05:
Try `var oldCode = ""+Storage.read(".bootrst")`

Also, just do `Storage.write(".bootbac",oldCode);` and don't specify an offset or length.

It could be related to ESP8266's unaligned byte read issues.

sirpy, 2018-06-05:
thanks.. adding ""+ solved it

closed 2018-06-05 by sirpy