espruino/Espruino #1458: http.get closes before end of data from v1.97

sirpy, 2018-06-05:
on ESP8266
I'm trying to download a 4k file via http.get
on v1.96 it works just fine, the res.on(data) is called multiple times and the whole file is downloaded
from v1.97-v1.99 res.on(data) is called only once with around 400bytes and then res.on(close) is called, so the file is not downloaded correctly.

gfwilliams, 2018-06-05:
@opichals ? Could this be related to the HTTP chunked stuff again?

opichals, 2018-06-05:
@gfwilliams Could be.
@sirpy Please provide a test case.

There is [test_http_post_chunked.js](https://github.com/espruino/Espruino/blob/master/tests/test_http_post_chunked.js) and other `test_http*.js` but perhaps that is not sufficient.

sirpy, 2018-06-05:
seems like it only happens on my nodejs server, it doesnt happen with local apache server.
The only thing i found different is that nodejs/expressjs uses lowercase response headers.
so if the http client checks for case sensitive headers it might be the issue.

opichals, 2018-06-06:
@sirpy Yeah, the HTTP headers letter case handling is not addressed.

Perhaps we could convert to lowercase [in the socketserver.c](https://github.com/espruino/Espruino/blob/master/libs/network/socketserver.c#L149) @gfwilliams ?

It would break existing header handling code though. An alternative lowercase headers lookup object could be added and later we could deprecate the current exact casing support.

gfwilliams, 2018-06-06:
... having said that about headers, maybe it's better to do it now rather than later :) I guess it's not too bad to change the code that uses it to `headers["Content-Type"]||headers["content-type"]`

opichals, 2018-06-06:
I'm sure we could find multiple use-cases for a case-insensitive Object lookup. I'd rather do that and allow users to use that from their JS code too (instead of doing the `jsvObjectGetChildIgnoreCase`).

gfwilliams, 2018-06-06:
> instead of doing the jsvObjectGetChildIgnoreCase

Not sure I understand?

> I'm sure we could find multiple use-cases for a case-insensitive Object lookup

Amazingly this doesn't seem to be part of basic JS. I guess we could expose it via an `E.lookupNoCase` sort of function?

opichals, 2018-06-06:
`E.lookupNoCase` sounds great!

gfwilliams, 2018-06-06:
@sirpy please can you try the latest travis builds out and see if this fixes your issue

closed 2018-06-06 by gfwilliams

sirpy, 2018-06-06:
i will... tonight. Thanks!

opichals, 2018-06-06:
I'd be nice to make the request/response headers objects actually be a Proxy as follows by default:
```
headers = new Proxy(headers, {
    get: (target, name) => E.lookupNoCase(target, name)
};
```

sirpy, 2018-06-06:
works great. thanks

commit 02aab33c (2018-06-06): Add E.lookupNoCase to allow searching case-insensitively for Object keys
Also Add robustness test for String.toUpper/etc in case they're bound to non-strings
Fix HTTP Chunked transfers when the server uses lowercase headers (fix #1458)