espruino/Espruino #1463: RFC: Updating the JS code from a HTTP(s) server

vshymanskyy, 2018-06-11:
I've been working on JS code OTA update, and I'd like to share some thoughts about this.
The idea is to host a js file on some server, then get the file, store it and run it somehow.

**Note:** This is a POC, and it was mostly tested on ESP32 only. I'm sharing this to raise the discussion, and possibly find a better solution.

### Storing JS file on the device
To store the file, we can either put it on `flashFatFS` / `SD card`, or in this case, the `Storage` module was used:
```js
let Storage = require('Storage');
Storage.write("main.js", `setInterval(function(){ console.log('This is sketch A') }, 5000)`)
```
An interesting option would be to replace the original `.varimg` file, but that requires generating it somehow on the host machine.

### Loading the js file on boot
Now, I'd like to run the script automatically on start. I found an easy way to do it:
```js
E.setBootCode(`var Storage = require('Storage'); Modules.addCached("App", Storage.read("main.js"))`)
```

After issuing `E.reboot()` the file is loaded and runs as expected.

### Changing the file in runtime
This one is also easy:
```js
Storage.write("main.js", `setInterval(function(){ console.log('This is sketch B') }, 5000)`)
E.reboot()
```

Sketch B runs.

### Generating a JS bundle
We also need a way to bundle all of the modules into a single JS file
For this, I created a `sketchA.js`:
```js
function draw(){
    g.drawString("Sketch A", 2, 2);
    g.flip();
}
I2C1.setup({ scl:22, sda:21 });
var g = require("SSD1306").connect(I2C1, draw);
```

We can bundle/minify it with:
```sh
espruino -n -m sketchA.js -o bundleA.js
```
`bundleA.js` looks like this:
```js
function start(){g.setFontVector(8),g.drawString('Hello Espruino!',2,2),g.flip();}Modules.addCached('SSD1306','function k(b){b&&(b.height&&(g[4]=b.height-1,g[15]=64==b.height?18:2,h[5]=(b.height>>3)-1),void 0!==b.contrast&&(g[17]=b.contrast))}var g=new Uint8Array([174,213,128,168,63,211,0,64,141,20,32,0,161,200,218,18,129,207,217,241,219,64,164,166,175]),h=[33,0,127,34,0,7];exports.connect=function(b,d,e){k(e);var f=Graphics.createArrayBuffer(128,g[4]+1,1,{vertical_byte:!0}),c=60;e&&(e.address&&(c=e.address),e.rst&&digitalPulse(e.rst,0,10));setTimeout(function(){g.forEach(function(a){b.writeTo(c,\n[0,a])})},50);void 0!==d&&setTimeout(d,100);f.flip=function(){h.forEach(function(a){b.writeTo(c,[0,a])});var a=new Uint8Array(129);a[0]=64;for(var d=0;d<this.buffer.length;d+=128)a.set(new Uint8Array(this.buffer,d,128),1),b.writeTo(c,a)};f.setContrast=function(a){b.writeTo(c,0,129,a)};f.off=function(){b.writeTo(c,0,174)};f.on=function(){b.writeTo(c,0,175)};return f};exports.connectSPI=function(b,d,e,f,c){k(c);var a=c?c.cs:void 0;c=Graphics.createArrayBuffer(128,g[4]+1,1,{vertical_byte:!0});e&&digitalPulse(e,\n0,10);setTimeout(function(){a&&digitalWrite(a,0);digitalWrite(d,0);b.write(g);digitalWrite(d,1);a&&digitalWrite(a,10);void 0!==f&&setTimeout(f,10)},50);c.flip=function(){a&&digitalWrite(a,0);digitalWrite(d,0);b.write(h);digitalWrite(d,1);b.write(this.buffer);a&&digitalWrite(a,1)};c.setContrast=function(c){a&&a.reset();b.write(129,c,d);a&&a.set()};c.off=function(){a&&a.reset();b.write(174,d);a&&a.set()};c.on=function(){a&&a.reset();b.write(175,d);a&&a.set()};return c}'),I2C1.setup({scl:22,sda:21});var g=require('SSD1306').connect(I2C1,start);
```

### Getting the file from HTTP server

[...in progress...]

### Open questions
- More efficient ways to store/load/run the file
- Compatibility with Espruino fw OTA updates (no issue with plain JS files, but for `.varimg` could be an issue)

gfwilliams, 2018-06-11:
How about just saving it as `Storage.write(".bootcde"` - this is basically the same as calling `E.setBootCode` and would save having to have the bootloader part that you made? It's basically just like doing 'save on send'.

You could also use some of the more advanced `Storage.write` functions to allow you to write to Storage in chunks (so you don't have to get all the program data in RAM before writing it).

vshymanskyy, 2018-06-11:
Ok, the problem might be in the way I put the file onto the Storage. Let me finish the HTTP(s) part and we'll see if it works.

gfwilliams, 2018-06-11:
Hmm. Looks like you're not quoting properly? Uploading like that you'd be better off adding the `--config MODULE_AS_FUNCTION=true` command-line option too. It'll save a lot of RAM once uploaded.

vshymanskyy, 2018-06-11:
@gfwilliams `MODULE_AS_FUNCTION=true` also solves my issue. The code runs now.
That's a bit weird as I'm sure the quoting was alright. Maybe the string was too long?
I have updated the OP. Next step: HTTP(s) ;)

vshymanskyy, 2018-06-11:
@gfwilliams I updated the HTTP(s) section
---
In general this method works quite well.
I'm experiencing some issues, for example ESP32 board freezes sometimes during download.
Also, I was unable to use the "writing with offset" API of `Storage`, as it always complains that the file is too small to store more data. I believe I initialized it correctly, as the first call was `f.write(".bootcde","",0,content_length+10)`

vshymanskyy, 2018-06-11:
Also, in some significant ~5% of runs, the OTA fails with:
```
Uncaught SyntaxError: Got ?[255] expected EOF
 at line 1 col 64
...?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255]
```.
If looks like it tries to load a clear page of flash.
Maybe some additional handling is needed, before calling `ESP32.reboot()`?

vshymanskyy, 2018-06-11:
In v1.99.174, I no longer see the `255] ?[255] ?[255] ?[255] ` issue. But it regularly gets stuck on downloading. The board gets completely unresponsive. Probably some ESP32 port issue.

vshymanskyy, 2018-06-12:
@gfwilliams I filed some issues separately.

Here's one specific to OTA functionality. Occasionally, it fails with: 
```log
IP: 192.168.1.67
Size: 5352
Got 5352 bytes
Uncaught SyntaxError: Got ?[255] expected EOF
 at line 3 col 26
      while(true) { E.reb ?[255] ?[255] ?[255] ?[255] ?[255] ?[255] ?[255]
                                                                   ^
in function called from system
```

The interesting thing is, this happens before the board has rebooted.
So it looks like the current sketch that is running, runs directly from flash, and when we start manipulating the `Storage`, it gets corrupted and we cannot finish the OTA process.

What do you think?