espruino/Espruino #1464: Any way to get the list of built-in modules?

vshymanskyy, 2018-06-12:


wilberforce, 2018-06-12:
The web ide does It - look at the console after connect.  

`console.log(process)

For this kind of question, please using the forums rather than raising as an issue - as it's not an issue with the firmware/build.

vshymanskyy, 2018-06-12:
Great. Thanks, I found it: 
```
process.env.MODULES.split(',')
```
No, i still think this is is issue - but more on documentation side.
I think this is worth adding to the API reference.

gfwilliams, 2018-06-13:
As @wilberforce says, please can you ask these kinds of questions on the forum in future?

So you want to add better descriptions to the `process.env` docs? They definitely seem out of date.

If you want to issue a PR it's as easy as clicking on `=>` next to the the method/field name in the docs and it'll bring you to the place in the code file in GitHub that needs changing.

PR #1469 (2018-06-13): Add docs on how to get the list of built-in modules
Fix #1464
closed 2018-06-13 by vshymanskyy