espruino/Espruino #1465: ESP32 Storage chunked write: Too much data for file size

vshymanskyy, 2018-06-12:
While working #1463, I faced this issue with ESP32 (didn't try on other boards).

```js
    max_len = 5015
    Storage.erase(fn);
    Storage.write(fn, "", 0, max_len);
    Storage.write(fn, data, offset);        // works
    Storage.write(fn, data2, offset2);    // fails
```

```log
Downloading: 5015
Chunk: 16 @ 0
Chunk: 536 @ 16
Uncaught Error: Too much data for file size
 at line 3 col 37
      Storage.write(fn, data, offset);

>process.env
={ 
  "VERSION": "1v99.174", 
  "GIT_COMMIT": "51ad06b", 
  "BOARD": "ESP32", 
  "FLASH": 0, "RAM": 524288, 
  "SERIAL": "d8a01d40-4920", 
  "CONSOLE": "Serial1", 
  "MODULES": "Flash,Storage,fs," ... "r,crypto,neopixel", 
  "EXPTR": 1073485300 }
>
```

wilberforce, 2018-06-13:
Please do a `reset(1)` to erase everything and re-try.

If you still have any issue, please provide a simple test case.

vshymanskyy, 2018-06-13:
OK, still not working. One more example:
```
var f = require("Storage");
f.write("a"," ",0,14);
f.write("a","Hello",0);
f.write("a"," ",5);              // <- Error: Too much data for file size
f.write("a","World!!!",6);
print(f.read("a"));
```

wilberforce, 2018-06-13:
Confirmed on Linux build.

The error is in the documentation, rather than the C implementation!

`f.write("a","Hello",0); `

Hello can't get written at position 0 - there is already as space there.

The write code checks that \xFF is in the slot before overwriting.

getting rid of the prints - you can see the character positions by dumping `read` - here is the corrected code:

```
var f = require("Storage");
f.erase("a");
f.write("a"," ",0,15);
f.read("a");
f.write("a","Hello",1);
f.read("a");
f.write("a"," ",6,1);
f.read("a");
f.write("a","World!!!",7);
f.read("a");
console.log(f.read("a"));
```

and the output:

>var f = require("Storage");
=function () { [native code] }
=undefined
>f.write("a"," ",0,15);
=true
=" \xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF"
=true
=" Hello\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF"
>f.write("a"," ",6,1);
=true
=" Hello \xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF"
=true
=" Hello World!!!"
>console.log(f.read("a"));
 Hello World!!!
=undefined

gfwilliams, 2018-06-13:
So the only issue here is really: 

```
f.write('.bootcde', "", 0, 100);  // <- fails to create an empty file
```

And the need for a note that you can't overwrite data that has already been written

vshymanskyy, 2018-06-13:
@gfwilliams agree, that would resolve all issues.

wilberforce, 2018-06-13:
`f.append("a",new_stuff)` would mean that you don't have to keep track of the last index used.

> Currently it returns true, and the file gets overwritten/truncated.
Are you sure?

this is working for me:

```
var f = require("Storage");
f.write("b","@",0,15);
f.read("b");
f.write("b","Fish",0); 
f.read("b");
f.write("b","Chips",0); 
f.read("b");
f.write("b","01234567890123",0); 
f.read("b");
```

="@\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF\xFF"
>f.write("b","Fish",0);
=true
="Fish"
=true
>f.read("b");
="Chips"
=true
>f.read("b");
="01234567890123"