espruino/Espruino #1466: ESP32 http.get freezes the board

vshymanskyy, 2018-06-12:
While working #1463, I faced this issue with ESP32 (didn't try on other boards).

Sketch to reproduce:
```js
var url = "http://vsh.pp.ua/OTA/espruinoA.js";

function httpLoop()
{
  console.log(`HTTP GET ${url}`)
  require('http').get(url, function(res) {
    var script = "";
    var len = res.headers['Content-Length'];
    console.log('Content-Length:', len);
    res.on('data', function(data) {
      script += data;
    });
    res.on('close', function() {
      console.log('OK');
      setTimeout(httpLoop, 1000);
    });
  }).on('error', function(e) {
    console.log("ERROR", e);
  });
}

var config = require('Storage').readJSON('.config');

if (config) {
  console.log(`Connecting to ${config.ssid}...`);
  var wifi = require('Wifi');
  wifi.connect(config.ssid, { password : config.pass }, function(err) {
    console.log("IP:", wifi.getIP().ip);
    httpLoop();
  });
}
```

Results in:
```log
WARNING: Bluetooth is disabled per ESP32.enableBLE(false)
WARNING: has simple connection not implemented yet
Connecting to structure7...
IP: 192.168.1.67
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5015
OK
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5015
[board freezes here]
```
This usually happens on the second iteration.

vshymanskyy, 2018-06-12:
Interestingly, the issue stops reproducing if switch to https (adding single letter in the `url`)!

wilberforce, 2018-06-13:
I have my wifi config saved, upload this:
```
var url = "http://vsh.pp.ua/OTA/espruinoA.js";

function httpLoop()
{
  console.log(`HTTP GET ${url}`);
  require('http').get(url, function(res) {
    var script = "";
    var len = res.headers['Content-Length'];
    console.log('Content-Length:', len);
    res.on('data', function(data) {
      script += data;
    });
    res.on('close', function() {
      console.log(`Got ${script.length} bytes`);
      setTimeout(httpLoop, 1000);
    });
  }).on('error', function(e) {
    console.log("ERROR", e);
  });
}

httpLoop();

```

output:

```
WARNING: gap set scan error code = 103
WARNING: set rssi scan not implemeted yet
WARNING: has central connection not implemented yet
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 1v99.40 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
=undefined
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
Content-Length: 5337
Got 5337 bytes
HTTP GET http://vsh.pp.ua/OTA/espruinoA.js
```

I think you need to do a 'reset(1)` to clear your flash, as there seems to be issues with your flash content.

vshymanskyy, 2018-06-13:
@wilberforce thanks, let me also try with another board.

wilberforce, 2018-06-13:
Everytime you update with a new firmware build, it's a good idea to use `reset(1)` so that anything that is in flash that might referencing addresses in the old firmware gets wiped out...

vshymanskyy, 2018-06-13:
@wilberforce can't reproduce now on another board. I'm pretty sure it wasn't `reset(1)` - related, as I didn't switch firmware versions.
Closing for now.

closed 2018-06-13 by vshymanskyy