espruino/Espruino #1477: ESP32: Slow SPI

vshymanskyy, 2018-07-05:
I just faced the same performance issue described here:
http://forum.espruino.com/conversations/321375
@gfwilliams commented that:
```
Personally I'd like to sort out SPI properly so that
it could work with buffers and DMA. That'd massively
improve performance on all platforms.
```
So I decided to create this ticked to track this issue.

gfwilliams, 2018-07-06:
Please don't open bugs like this. This is already more than covered in:

* https://github.com/espruino/Espruino/issues/695
* https://github.com/espruino/Espruino/issues/1212

Last thing I need is 50 of every bug, one for each board.

closed 2018-07-06 by gfwilliams

vshymanskyy, 2018-07-06:
Sorry, both these issues seemed unrelated at first glance