espruino/Espruino #1478: ESP32: PSRAM support?

vshymanskyy, 2018-07-05:
Many esp32 modules come with 4Mb PSRAM now, using that could greatly improve Espruino capabilities! Would it be hard to provide a PSRAM-enabled build?
ESP-IDF 3.0 should have the needed switches already: http://esp-idf.readthedocs.io/en/latest/api-guides/external-ram.html

wilberforce, 2018-07-05:
> ESP-IDF 3.0 should have the needed switches already: http://esp-idf.readthedocs.io/en/latest/api-guides/external-ram.html

With the 3.0 release there was no way of having a single binary that supported PSRAM or without. It had to be selected at compile time. 3.1 beta supports this, and @jumjum123 has been preparing for this support. When 3.1 is realeased we can look at adding to the build so that if PSRAM is available, it will be used.

wilberforce, 2018-07-05:
And as esp-idf is a moving target, there are other knock on effects to be solved, see the comments here:
http://forum.espruino.com/conversations/319948/?offset=25

vshymanskyy, 2018-07-05:
@wilberforce thanks!