espruino/Espruino #1479: Superclass functions are behaving as getters when called from subclass

Ignigena, 2018-07-06:
When calling a superclass function I am getting the error: `Uncaught Error: Expecting a function to call, got [String|Object]` when the superclass function has a return value and `Uncaught Error: Function not found!` when there is no return value.

Using the example in the tests:

```
class Cat { 
  constructor(name) {
    this.name = name;
  }  
  speak() {
    return this.name + ' makes a noise.';
  }  
}

class Lion extends Cat {
  speak() {
    return super.speak()+this.name + ' roars.';
    // Uncaught Error: Expecting a function to call, got String
  }
}

var animal = new Lion('Nigel');
animal.speak();
```

In this case `super.speak` is not a function as expected but is instead a string containing the return value of the parent. It seems to be acting as if `speak` was declared with `get speak()` on the parent.

If the parent has no return value, the entire function fails.

```
class Cat { 
  constructor(name) {
    this.name = name;
    this.timesSpoken = 0;
  }  
  meow() {
    this.timesSpoken++;
  }  
}

class Lion extends Cat {
  roar() {
    super.meow();
    return this;
    // Uncaught Error: Function "meow" not found!
  }
}

var lion = new Lion('Nigel');
lion.roar().roar();
console.log(lion.name + ' roared ' + lion.timesSpoken + ' times!');
```

Ignigena, 2018-07-06:
Just realized this doesn't seem to be limited to just calling `super` on a class method. Seeing the same behavior anytime I return a string in a method on a class whether or not it is subclassed:

```
class Cat { 
  constructor(name) {
    this.name = name;
  }
  speak() {
    return this.name + ' meows.';
  }
}

var animal = new Cat('Nigel');
console.log(typeof animal.speak, animal.speak); // Returns "string Nigel meows"
console.log(animal.speak()); // Uncaught Error: Expecting a function to call, got String
```

gfwilliams, 2018-07-09:
Thanks for the report - What device are you running on, and which version of Espruino?

I just tried this on a current Linux build, and it looks ok:

```
>class Cat { 
:  constructor(name) {
:    this.name = name;
:  }
:  speak() {
:    return this.name + ' meows.';
:  }
:}
=function (name) { ... }
>var animal = new Cat('Nigel');
=Cat: { 
  "name": "Nigel"
 }
>console.log(typeof animal.speak, animal.speak);
function function () {return this.name + ' meows.';}
=undefined
>console.log(animal.speak());
Nigel meows.
```

Ignigena, 2018-07-09:
I am running on your MDBT42Q module, and the firmware was on the nightly build of Espruino (1v99.171) since I was trying to test the new getters and setters feature.

gfwilliams, 2018-07-10:
Perfect, thanks for this - I've just tried and I get exactly the same. It helps a massive amount to have such a clear test case.

This one's going to be fun to track down as it doesn't happen on Linux builds - even when supposedly building with exactly the same settings as the microcontroller.

closed 2018-07-10 by gfwilliams

Ignigena, 2018-07-10:
You are awesome! Thanks for resolving so quickly!

commit cc22e5f1 (2018-07-10): Fix regression parsing methods in classes on embedded (fix #1479)