espruino/Espruino #1480: ESP32: I2C access on wrong address

vshymanskyy, 2018-07-08:
When reading on wrong address operation success, returns random data.

I think it should actually throw an error. Currently it only prints to the console:
```
ERROR: jshI2CRead:, slave doesn't ACK the transfer.
```
This also applies to `writeTo`

MaBecker, 2020-03-13:
@vshymanskyy do you like to create a pull request for this?

MaBecker, 2020-08-30:
just tested - it has been fixed.

```JS

I2C1.setup({sda:D12,scl:D22});

try{
  I2C1.readFrom(0x20,1);
} catch(e){
  console.log("catch:",e.message, e.stack);
}
/* output

>catch: jshI2CRead: slave doesn't ACK the transfer  at line 6 col 23
  I2C1.readFrom(0x20,1);
                      ^
*/
```

closed 2020-08-30 by MaBecker