espruino/Espruino #1481: WebSocket Module possible error

plafhz, 2018-07-17:
Hi friends,

I'm doing a project with espruino that use websocket to connect to an Django server, I'm using Django channels to do it. When I tried to connect them, in Wireshark I capture the package sended by the esp8266 and the responses by de Django server, i get an http 400 error because the header field Sec-WebSocket-Key must be an 24 bytes long, insted ws module send a string with 16 bytes long.

[image]

To solve this I go to ws.js module source code ([http://www.espruino.com/modules/ws.js](url)).
I found the error in the function buildKey()

`
function buildKey() {
    var randomString = btoa(Math.random().toString(36).substr(2, 18));
    var toHash = randomString + '258EAFA5-E914-47DA-95CA-C5AB0DC85B11';
    return {
        source: randomString,
        hashed: btoa(require('crypto').SHA1(toHash))
    }
}`

Here the randomString is 16 bytes long that produce the error in the server.

So I change it for:

`
function crearRandomString(length){
  var str = "";
  for ( ; str.length < length; str += Math.random().toString( 36 ).substr( 2 ) );
  return str.substr( 0, length );
}
function buildKey() {
  var randomString = btoa(crearRandomString(16))
  var toHash = randomString + '258EAFA5-E914-47DA-95CA-C5AB0DC85B11';
  return {
    source: randomString,
    hashed: btoa(require('crypto').SHA1(toHash))
  }
}
`

and this work fine.

The RFC6455 have an example of this field and have an string of 24 bytes long.

So I have two questions:

first: Is ther any reason for the 16 bytes long, maybe other system works like that?

second: Where are the modules source codes of espruino? I want to send a pull request buy I don't know where.

Sorry if this is not the best place to post the Issue. I would be grateful if you can correct me in something if I'm wrong.

Thanks.

wilberforce, 2018-07-17:
@gfwilliams 

https://tools.ietf.org/html/rfc6455#page-18

7.   The request MUST include a header field with the name
        |Sec-WebSocket-Key|.  The value of this header field MUST be a
        nonce consisting of a randomly selected 16-byte value that has
        been base64-encoded (see Section 4 of [RFC4648]).  The nonce
        MUST be selected randomly for each connection.

The string is built with:
`var randomString = btoa(Math.random().toString(36).substr(2, 18));`

breaking this down:
`Math.random().toString(36).substr(2, 18);`

generates a 11 byte string as the number of digits are not enough.

```
>Math.random().toString(36).substr(2, 18).length
=11
```

Here is a quick fix, creating 2 8 byte chunks:

```
var randomString = btoa(Math.random().toString(36).substr(2, 8)+Math.random().toString(36).substr(2, 8));
="OXA3d3d3aXo4cDFzbnVqdjU4MG8="
```

This is quite verbose - so this uses adds on `E.hwRand()` to add the 5 more bytes..

```
>var randomString =btoa(((Math.random().toString(36)+''+E.hwRand()).substr(2, 16)));
="Y2hjZHh1cWQwbGMyMjg1OQ=="
```

@2plafhz

The source for the module is here:

https://github.com/espruino/EspruinoDocs/blob/master/modules/ws.js

gfwilliams, 2018-07-17:
Thanks for such an in-depth bug report - I just added @wilberforce's fix (hwRand isn't available on some devices, so I used the first option).

As he says, all the JS modules are found in https://github.com/espruino/EspruinoDocs - so if you have other fixes you can just issue a bug or PR there and I can pull in the changes.

thanks!