espruino/Espruino #1484: ESP32: Neopixel does not work anymore with experimental 1v99

NielsLeenheer, 2018-08-06:
When using the experimental 1v99 version of Espruino, using the following command from the document to change the color of a NeoPixel nothing happens: 

````is
require("neopixel").write(D13, [255,0,0]);
````

However, if I flash the current 1v95, it works without issues.

wilberforce, 2018-08-07:
Could this be related to ?

https://github.com/espruino/Espruino/commit/fcccdccb4a826e4ce66c6dea1214a4100acad5a2

gfwilliams, 2018-08-07:
Nope - ESP32 uses its own function esp32_neopixelWrite - which as far as I know hasn't changed since it was created ages ago.

I guess it's the new ESP32 SDK?

wilberforce, 2018-08-19:
v1_98 is the last working version
https://www.espruino.com/binaries/espruino_1v98_esp32/
2018-05-18

wilberforce, 2018-08-19:
@jumjum123 

This has stopped working from we changed to the V3.0 sdk.

Any ideas of what needs to be updated here:

https://github.com/espruino/Espruino/blob/master/targets/esp32/esp32_neopixel.c

?

jumjum123, 2018-08-21:
Hmm, I only know about a change around V3.1
  SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_RMT_CLK_EN);
  CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_RMT_RST);
needs to be replaced with
  DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_RMT_CLK_EN);
  DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_RMT_RST);
To be honest, I did not check. And for about a month I will not have much time.

wilberforce, 2018-12-16:
This has been fixed in 
https://github.com/espruino/Espruino/pull/1585

closed 2018-12-16 by wilberforce