espruino/Espruino #1487: Uncontrolled recursion in matchhere results in stack overflow.

dominiakm, 2018-08-09:
Version:
05b7dd2fe45a919157085e92f8fcb9a48749d3e0 

OS:
Ubuntu 16.04

Testcase:
An inline python script generating the real testcase:
```
python -c "print 'var x = new RegExp(\'' + '(' * 23000 + '\')'; print 'x.exec(\'something\')'" > ./test ; ./espruino ./test
```
The actual testcase has the following form:
```
var x = new RegExp('(((...(')
x.exec("something")
```

Result:
Segmentation fault.
Stack overflow due to too deep recursion at src/jswrap_regexp.c:175

Note that because this crash occurs due to stack space exhaustion this exact testcase may not work on all systems.

gfwilliams, 2018-08-09:
Thanks - this could probably be fixed using some of the same stack checking routines used in `jsparse.c`.

IMO this isn't so urgent though as it'd only be a vulnerability if you were executing a RegEx from an untrusted source.

dominiakm, 2018-08-09:
I agree with both statements ;)

commit 7ef769f4 (2018-09-03): Fix stack overflow if executing regex full of hundreds of open brackets (fix #1487)
closed 2018-09-03 by gfwilliams