espruino/Espruino #1489: small numbers a convert to zero if smaller than 1e-9, big numbers are strip to 1e68 when convert to string

jbgagnon, 2018-08-10:
so we can't do any physics ...

to test:
console.log(1e-9) write 0
'x:" + 1e-9 write "x:0"
JSON.stringify({x:1e-9}) return ="{\"a\":0}"
Math.pow(2,-32) says 0
('' + 1e72).length says 69
JSON.stringify({x:1e80}) is also wrong so can't be communicate to an external service
console.log(1e100/3) write something not enough long ( or two long 3.333333333333333e+99 is the good answer)

the bug is in
file: jsutils.c
function: ftoa_bounded_extra

it must test if exponent must be added for example if 
val < 1e-9 or val > 1e64 (I don't really know the actual limits)
but double value have  17 chars of precision so probably test val > 1e17 (it also depends of buffer length, normaly a buffer of 32 chars is good (sign + dot + 17 chars + e + sign + exponent max 3 char  = 24 chars)

bug probably or some platform (esp32, esp8266, ...) we can use native library and smaller this code ... no ?

gfwilliams, 2018-08-10:
I think parsing floating point numbers works just fine. `0==1e-20` is `false`. It's just converting to Strings.

Custom double-to-string code was originally added because platform-specific versions pulled in large amounts of extra code - on some platforms it caused 1kB of RAM to be statically allocated. I believe JS may have some standards on how double to string code should behave as well (not that we currently adhere to those).

So I'd like to keep the implementation inside Espruino if only because it aids testing to always use the same implementation - but you're welcome to contribute an updated `ftoa_bounded_extra` function that'll use `e` for larger/smaller values if you want to.

jbgagnon, 2018-08-10:
hello Gordon,
I go en hollidays tomorrow. But when i come back at end of august i will be
happy to contribute to this project.
JB Gagnon

gfwilliams, 2018-10-25:
Any chance of doing this?

jbgagnon, 2018-10-25:
> Yes - best to ask on the forum. You'll have much better luck with Windows
> 10 Linux Subsystem or with a virtual machine running Linux though

Ok I can compile now !
(making some change in provision.sh for path to access get_board_info.py,
and do a pacman for tar command ....)

so to train myself I correct this issue (the easy one)
https://github.com/espruino/Espruino/issues/1488

I test it on my board, it's ok ! no "back to the future" at all

what is the good way to proceed (is-there a guide ?)
must a need to create a branch ?
must somebody make a code review before commiting ?
or just commit and make a pray to not have done a stupid mistake (newbe
mistake)
(I develop since many many year, but ... (we use svn not git))

JBG

gfwilliams, 2020-10-29:
Fixed with https://github.com/espruino/Espruino/commit/1e6586a80cd61063a11f63bb026ac5db35735d53

closed 2020-10-29 by gfwilliams