espruino/Espruino #1495: integration of analog pin to use Waveform

MaBecker, 2018-09-01:
Analog read is implemented on ESP8266 and works without parameter as there is only one analog input. Pins are define D0 to D16, so A0 is missing for now. Like to change this so it is possible to use Waveform()  to read multiple analog values with one call to pass it to a median filter or even combine it into one.

I would be glad about any note to get this coded for ESP8266 - thanks.

@wilberforce  and @jumjum123  is this relevant for ESP32 too?

gfwilliams, 2018-09-03:
You can basically just add it in the python file. You just need an extra item in the 'pins' array: https://github.com/espruino/Espruino/blob/master/boards/PIXLJS.py#L158

Stick it at the end, so `D0` stays mapped to the actual integer `0`.

**However** I can't quite remember how the util timer works on ESP8266, but if it doesn't use interrupts then you might struggle to use Waveforms... Can you play back ok?

MaBecker, 2018-09-03:
Thanks, ok now it shows the A0 with ACD, 

[image]

A0 added as last pin which is pin 17

     { "name":"PA0", "sortingname":"A00",  "port":"A", "num":"17", "functions":{ "ADC1_IN0":0 }, "csv":{}

Just tried `digitalWrite(A0, 1)` which  cause a reset, have to catch this in ESP8266 implementation

Added this to reject calls for A0.

    /* reject analog pin 17 */
    if (pin == 17 ) { 
      jsError("not possible for analog pins");
      // or jsExceptionHere(("not possible for analog pins");
      return;
    }

**Is there a smarter way to reject?**

>However I can't quite remember how the util timer works on ESP8266, but if it doesn't use interrupts then you might struggle to use Waveforms... Can you play back ok?

Well after thinking about it, skipped to use Waveform() because pwm is too poor on ESP8266.

gfwilliams, 2018-09-04:
In read/write/etc I'd just ignore anything that doesn't have the `pinInfo[pin].port` set to Port D. That's nice and flexible :)

MaBecker, 2018-09-04:
Thanks, so I will use something like this:

    if ((pinInfo[pin].port & JSH_PORT_MASK) == JSH_PORTA) { 
        ....
    }

or

    if ((pinInfo[pin].port & JSH_PORT_MASK) != JSH_PORTD) { 
        ....
    }

MaBecker, 2018-09-04:
A0.getInfo()
={
  port: "A",
  num: 17, channel: 0,
  functions: {  }
 }
A0.getMode()
="analog"

commit 0080541c (2018-09-05): fully integration of analog pin A0 (fix #1495)
closed 2018-09-05 by gfwilliams