espruino/Espruino #1506: arraybuffer dont works on ESP32

jbgagnon, 2018-09-11:
a = new Int32Array(1);
a[0] = 5; // print 0 not 5
console.log(a[0]); // print 0 not 5

it works in the 2018-08-31 version
it dont works in the 2018-09-05 version

gfwilliams, 2018-09-12:
Pretty sure this works on other platforms.

Is this using travis builds, or have you built it yourself? If you built yourself, did you `make clean`?

Probably related to https://github.com/espruino/Espruino/commit/6b1b9f5f2d43c5b7d1ba36edf19f399ad0c3ccef but not sure why since for Int32Array the code executed should be identical.

gfwilliams, 2018-09-12:
Don't worry about compiling - it's the Travis builds that are useful for figuring out what works and what doesn't

jumjum123, 2018-09-21:
This is an issue for Int32Array and Uint32Array on ESP32. All other arrays are working fine.
Is there any way to peek from Javascript for data in IntArrays ?
I tried to use E.getAddressOf and E.dumpLockedVars, but get no idea what the returned data is.
Creating an array with *jswrap_arraybuffer_constructor looks fine (added some fprints ;-).

gfwilliams, 2018-09-21:
It should work, but you have to have an array big enough that Espruino thinks it's worth creating a flat string for:

```
>var a = new Uint32Array([1,2,3,4,5,6,7,8,9,10,11,12,13,14]) 
=new Uint32Array([1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14])
>var b = E.getAddressOf(a,true)
=37015328
>peek32(b)
=1
>peek32(b+4) 
=2
```

It'd be worth seeing if it works before https://github.com/espruino/Espruino/commit/6b1b9f5f2d43c5b7d1ba36edf19f399ad0c3ccef and not after it. It *shouldn't* have changed anything, because the 32 bit case should be identical - but I guess I may have missed something.

gfwilliams, 2018-09-21:
Thanks - that helps to narrow it down? Does it work for float arrays?

jsvArrayBufferIteratorDataToInt might be a good place to start - you can see if the data coming in is all 0s, and if it comes out as 0 as well :)

jumjum123, 2018-09-21:
It works fine with float32 and float64.
Only Int32Array and Uint32Array have the problem.
I'll check jsvArrayBufferIteratorDataToInt

gfwilliams, 2018-09-21:
Actually I think it could be `JsVarInt mask = (JsVarInt)((1UL << bits)-1);` - it's possible the compiler works differently on ESP32 and when `bits==32` mask becomes 0 because of overflow.

The old code (before https://github.com/espruino/Espruino/commit/6b1b9f5f2d43c5b7d1ba36edf19f399ad0c3ccef) actually sneakily worked around that issue, so that could be what you're seeing

jumjum123, 2018-09-21:
I've added in 
printf("dataLen:%d,bits:%d,mask:%d,v:%d\n",dataLen,bits,mask,v);
printf("mask:%d,%d,%d\n",1UL,(1UL << bits),(1UL << bits)-1);
and got this 
dataLen:4,bits:32,mask:0,v:76
mask:1,1,0
where 76 would be the correct value, but mask looks strange to me.
BTW, I get these 2 lines twice, looks like function is called twice ???

gfwilliams, 2018-09-21:
Can you try that please?

jumjum123, 2018-09-21:
Tested and works fine now with Javascript:
**var a = new Uint32Array([64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80]);
console.log(a[0]);**
And this change to jsvArrayBufferIteratorDataToInt
**static JsVarInt jsvArrayBufferIteratorDataToInt(JsvArrayBufferIterator *it, char *data) {
  unsigned int dataLen = JSV_ARRAYBUFFER_GET_SIZE(it->type);
  unsigned int bits = 8*dataLen;
  JsVarInt mask = (JsVarInt)((1ULL << bits)-1);
  JsVarInt v = *(int*)data;
printf("dataLen:%d,bits:%d,mask:%d,v:%d\n",dataLen,bits,mask,v);
printf("mask:%d,%d,%d\n",1UL,(1UL << bits),(1UL << bits)-1);
  v = v & mask;
  if (JSV_ARRAYBUFFER_IS_SIGNED(it->type) && (v&(JsVarInt)(1UL<<(bits-1))))
    v |= ~mask;
  return v;
}**
In console I get this, the twice effect
**dataLen:4,bits:32,mask:-1,v:64
mask:1,1,0
dataLen:4,bits:32,mask:-1,v:64
mask:1,1,0
64**
Didn't dig deeper into this (yet)

jbgagnon, 2018-09-21:
on travis the espruino_esp32.bin | 2018-09-21 14:33 | 1.1M | 
works fine !
thank you !

gfwilliams, 2018-09-21:
Great!

And @jumjum123  I just checked that second read and it was to do with the code for getters/setters. I believe I have managed to modify the code to avoid doing that now.

closed 2018-09-21 by gfwilliams

jumjum123, 2018-09-21:
Three guys are working together on a problem at the same time and get it done and checked.
I love it.
Wish you guys a nice weekend, mine will start now.