espruino/Espruino #1512: Module AT.js: getData() returns garbage '\n' character

wklenk, 2018-09-20:
When dealing with the Quectel BG96 radio model, common practice is to register for the unsolicited return code `+QIURC` using the AT module. As soon as this URC is received, a callback function is called that is used to read out the actual data from the peer.

An example URC might look as follows:
`\r\n+QIURC: "recv",0,4\r\n1234`
This means, 4 bytes where received from the peer and the bytes are 1234

But if you call `at.getData()` to actually read out the data, you get:
`\n123` instead of `1234`

So there is obvioulsy a bug in the code that doesn't eliminate the garbage newline character '\n'.

I attach a JavaScript script that can be used to reproduce this bug. The output of this script is as follows:

```
js at_test.js 
] "\r\n+QIURC: \"recv\",0,4\r\n1234"
Got URC: "+QIURC: \"recv\",0,4"
[ '+QIURC: "recv"', '0', '4' ]
Received data from peer: "\n123"
] "4"
```
[at-bug-test.js.txt](https://github.com/espruino/Espruino/files/2400629/at-bug-test.js.txt)

gfwilliams, 2018-09-20:
Thanks - I think the issue here is that `registerLine` is being called after the `\r` character, because I believe at some point there were some AT command devices that just used `\r` rather than `\r\n`. While it'd normally remove the newline in anything it passed, because data is being grabbed right after execution it's passing along the newline.

It's tempting to change the module to use `\r\n`, but I'd be quite worried that such a big behaviour change would break other modules that depended on it.

Are you encountering this with the https://github.com/espruino/EspruinoDocs/blob/master/devices/QuectelBG96.js module?

It seems that you could probably work around it using:

```
 at.getData(1, function(newline) {
    at.getData(lineSplittedUp[2], function(data) {
      console.log("Received data from peer: " + JSON.stringify(data));
    });
  });
```

gfwilliams, 2018-09-20:
Actually, please can you wait a bit on this? I'll see about changing the AT module for `\r\n` and will do some testing - it'd simplify a lot of things.

closed 2018-09-21 by gfwilliams

gfwilliams, 2018-09-21:
Ok, just done. That should improve things for you I hope, although you may need to tweak the handling of ATE0 in your copy (see the changes in https://github.com/espruino/EspruinoDocs/commit/ebcb3048a21f491edb63f8a71b3c94b7f7a4d4b8)

Hopefully this doesn't cause problems with other AT-command based devices, but I guess we'll see!

As I mentioned in the forum, if you're happy to share it'd be great to be able to build your NB-IoT related changes into `QuectelBG96.js`

gfwilliams, 2018-09-24:
Thanks - the IDE *should* always use the new version - everything's set to work without caching (it should work).

The only issue is if you have enabled 'offline mode' in which case it'll always work from the cache of offline modules that it downloaded.