espruino/Espruino #1514: Module MQTT: Connect timeout (10s) not configurable

wklenk, 2018-09-24:
I currently try to implement a data logger that uses MQTT as protocol over a LTE Narrowband IoT (NB-IoT) communication line.  Unfortunately, the latency on this communication line is very high sometimes. 

The MQTT module expects a response to arrive at most 10 seconds (timeout) after the initial request was sent to the MQTT server (I am using Cayenne myDevices). If a response doesn't arrive, the MQTT client terminates the connection.

It would help if this timeout would be configurable via "options" like other MQTT settings.

gfwilliams, 2018-09-24:
Hi - In the future, please  can you make issues related to the Espruino modules in the https://github.com/espruino/EspruinoDocs repo, as that's where they reside?

Do you mean the initial connection timeout? https://github.com/espruino/EspruinoDocs/blob/master/modules/MQTT.js#L69

That is public so you should still be able to modify it...

full source code here if you need: https://github.com/espruino/EspruinoDocs/blob/master/modules/MQTT.js

wklenk, 2018-09-25:
> That is public so you should still be able to modify it...

Sorry, I did not see the setting when I did the "dump()" operation. Found out now that there is a way to hide things in the module implementation.

Thanks, it works now :)

closed 2018-09-25 by wklenk