espruino/Espruino #1517: Was there a reason for adding SHA256 and SHA512 on ESP8266_4MB

MaBecker, 2018-09-29:
As far as I know there is only module `ws` that is using sha1? Or do I miss some modules or functions 
that use SHA256 or even SHA512?

This is all about remove not needed stuff to have space for important things.

Like to remove it if not needed.

gfwilliams, 2018-09-29:
I haven't heard of anyone needing it. I think they just got added because there was space for them at one point.

Are we still trying to fit in 512k on the 4MB parts though?

gfwilliams, 2018-09-29:
Maybe leave this open for a week and see if anyone complains, then remove them?

hungryforcodes, 2018-09-29:
What would it normally be used for, aside from HTTPS related stuff? A version of TEA (Tiny Encryption Algorithm) might be nice to put in the ESP8266 build instead, if I had to make choices about space. Unless someone has another compact encryption library they would nominate instead.

MaBecker, 2018-09-30:
> Are we still trying to fit in 512k on the 4MB parts though?

Fortunately not :-)

> Maybe leave this open for a week and see if anyone complains, then remove them?

Yes

> A version of TEA (Tiny Encryption Algorithm)

check #1055

hungryforcodes, 2018-09-30:
Ah, I missed that for some reason -- thanks! ;)

MaBecker, 2018-10-12:
Ok, lets remove it

commit 918232c0 (2018-10-15): remove  SHA256 SHA512 (fix #1517)
commit eb75eea5 (2018-10-15): Merge pull request #1521 from espruino/MaBecker-patch-1

remove  SHA256 SHA512 (fix #1517)
closed 2018-10-15 by gfwilliams