espruino/Espruino #1532: Storage.readArrayBuffer do 1 variable memory leak

jbgagnon, 2018-10-19:
require('Storage').write('f','hello world !');
process.memory();
for(var i = 0; i< 100;i++) {
    var a = require('Storage').readArrayBuffer('f');
 }
 process.memory();

// you loose 100 variable !

jbgagnon, 2018-10-19:
same problem with readJSON

require('Storage').write('f',JSON.stringify({x:3,y:4}));
process.memory();
for(var i = 0; i< 100;i++) {
 var a = require('Storage').readJSON('f');
 }
 process.memory();
// you loose 100 varaibles again ! (need to reboot now ... )

iotong, 2018-10-20:
no！you need call process.memory() after readJSON or readArrayBuffrr

gfwilliams, 2018-10-20:
Thanks - yes, it looks like they're missing unlocks. Thanks for the report - I'll sort it on Monday

commit c4c63474 (2018-10-22): Fix Storage.readJSON/readArrayBuffer memory leak (fix #1532)
closed 2018-10-22 by gfwilliams