espruino/Espruino #1535: what speaks against  console baudrate 9600?

MaBecker, 2018-10-21:
Is there any reason why we should stick with default_console_baudrate of 115200 for ESP8266_BOARD and ESP8266_4MB?

gfwilliams, 2018-10-22:
I'm biased, but apart from the existing documentation out there saying 115200 I don't see a reason.

I mean, even with all the docs out there people routinely forget to set the baud rate anyway, and it looks like many people use Telnet anyway, which is faster.

With the recent storage changes it should be trivial for more advanced users to do `require("Storage").write('.boot0','Serial1.setup(115200,...)');` anyway if they really need it.

MaBecker, 2018-10-22:
@wilberforce  and @jumjum123  what about ESP32 and baud 9600?

Frida854, 2018-10-22:
What about PICO, it runs with 115200, even with latest version 2V00.
I'd be sad to let speed down.

jumjum123, 2018-10-22:
Looks like its grown the same way as ESP8266 ;-) Default from Espressif is 115200
AFAIK, we had to change default in menuconfig, ESP32.py and jshardwareUart.c
I would prefer to have higher speed, but thats my personal opinion only

May be we could add a command to save default rate for specific board somewhere in flash ?
Similiar to the option we have in ESP32 to switch Bluetooth or WiFi on/off
This would give anybody the option to use its own preferred baudrate

BTW, a quick search for 115200 in sources showed about 70 files

MaBecker, 2018-10-22:
Well, check CONSOLE log: 

```
>>> Connecting...
Set Slow Write = true
[object Object]
Connected [object Object]
Received a prompt after sending newline... good!
>>> Sending...
```

even if Throttle Send is not set

[image]

gfwilliams, 2018-10-23:
Interesting about the other ESP8266 stuff - however I guess it's reasonable to assume that most users will go straight to using the IDE or CLI? If you're using your own stuff you're probably more likely to understand about the baud rate?

@Frida854 it's fine on all the Espruino boards like Pico/WiFi/Original. The USB goes straight into the microcontroller, so actually it doesn't matter what you set the baud rate to - it'll work at the same (high) speed through USB regardless. This only really applies to ESP8266/ESP32 that have a separate USB chip.

@MaBecker yes, the IDE does just turn on Slow Write regardless. I guess there could be an IDE option to fully force it not to be used *BUT* if you did that, you'd find that some uploads randomly failed.

115200 would be great *if* there was flow control - but as there isn't we've got to throttle on the IDE side - add to that all the confusion from the different baud rates and I'm more inclined to say 9600 is the best bet.

closed 2019-03-11 by MaBecker

gfwilliams, 2019-03-11:
Looks to me like there isn't much opposition to choosing 9600 baud?