espruino/Espruino #1538: save() on nucleof401re 2v00 is broken.

MarkDavidOeltjenbruns, 2018-10-24:
I tried Espruino 2v00 on my nucleof401re board but found that when I would run save() it would lockup after Writing..... and the next line would only show the 'C' From the Compressed ... line.  I tried 1v99 and it works as expected.

gfwilliams, 2018-10-25:
Can you try `require("Storage").eraseAll()` in case it's something to do with bad data left in flash memory?

MarkDavidOeltjenbruns, 2018-10-25:
Forgot to mention, before posting I checked for other 'save doesn't work' type of issues and tried the suggestions out.  Just to be sure, I tried your suggestion just now and it still ends the same way.
So the eraseAll doesn't seem to be erasing everything, or something else is wrong.

I then went on to compile a debug version of 2.00, and get a repeating loop of:
ASSERT(strlen(name)<=8) FAILED AT src/jsflash.c:56
  #1[r1,l2] Object {
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l1] Object {
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r2,l0] Array(0) [ ]
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r2,l0] Array(0) [ ]
      }
  }
REBOOTING.Not loading saved code from different Espruino firmware.
After doing a bit more of the eraseall testing, It still is an infinite loop, but I don't seem to always seem to see the rebooting message even though it is clearly stuck in a reboot cycle, although I did see "Not loading saved code from different Espruino firmware." when I went back to the 2v00 official version so that REBOOTING line could just be when I wasn't completely 'clean.'  The ASSERT message is the same in both cases.

It does seem to be erasing correctly.  After a save, in 1v99 I do it twice, the first time it takes ~second and the next time almost no time.  
For completeness this is what I see when running 2v00 official release.
>var on = false;
=false
>save();
=undefined
Compacting Flash...
Calculating Size...
Writing..
C

MarkDavidOeltjenbruns, 2018-10-26:
It is now working as expected.

One minor thing I did notice was that although the BUILDNUMBER was updated correct (17-27 in my case) in the output name, it did not update the JS_VERSION and recompile the files that use it, like jsinteractive.c which shows it on startup.  Although you could fix this, it was easy to make clean first. Depends on how picky you are on your dependencies being correct. :-)

closed 2018-10-26 by MarkDavidOeltjenbruns