espruino/Espruino #1539: Storage can't rewrite many times a file with same name (like configuration save)

jbgagnon, 2018-10-25:
We have the error "File already written with different data" and the Storage is corrupt,
nothing can be save with another name
and require('Storage').compact() make reboot each time
and require('Storage').getFree() says plenty of free space (233456)
only possibility : erase and reflash ...

to test:
try {
  for(i=0;i<500;i++) {
    require('Storage').write('f', 'hello ' + i);
  }
} catch(ex) { console.log('ex', i, ex); }

perhaps depending the size of '.varimg' , the exception arrive near 50 iteration
so when I want to save application state, it arrive very rapidly !

( Is-it a problem of file across flash pages ?
I don't understand the test in sources jsflash.c
  addr += (uint32_t)offset;
  if (!jsfIsErased(addr, (uint32_t)dLen)) {
    jsExceptionHere(JSET_ERROR, "File already written with different data");
    return false;
  }
)

gfwilliams, 2018-10-25:
Which board is this on? 233k is a lot of storage!

You could do `Storage.eraseAll()` to recover.

`File already written with different data` is because you can allocate a larger file, and then write just parts of it - see http://www.espruino.com/Reference#l_Storage_write

I just found the issue I think - it was actually when creating the file it'd tried to create a new file over an older one

closed 2018-10-25 by gfwilliams

jbgagnon, 2018-10-25:
Hello Gordon

it use ESP32 (doit.am devkit v1)

about space :
1) and it-il possible to have a different partition table without OTA ? to
have more space
2) without flash card is-it possible to use flash between 0x300000 and
0x400000 (partition name storage (improter name because storage is on
js_code partition)) is a not use "fs" module ?

thank you

JB

jbgagnon, 2018-10-25:
I test the new version (espruino_2v00.20_esp32.bin ) : very good ! 
thanks a lot

gfwilliams, 2018-10-25:
You'd have to ask on the ESP32 Espruino forum about the other stuff: http://forum.espruino.com/microcosms/1086/

I can't afford to provide free support for ESP32 specific stuff I'm afraid - unless you're willing to pay for consultancy work :)

commit a3726919 (2018-10-25): Fix Storage write error when skipping pages (fix #1539)