espruino/Espruino #1544: Hope to introduce the nrf5 mesh network into espruino.

espirate, 2018-10-29:


gfwilliams, 2018-10-29:
You are going to do this yourself? Or you would like someone else to add it?

I've been looking at this for a while, but Bluetooth Mesh isn't as amazing as it sounds none of the devices acting as internal mesh nodes are low power at all, so it's not really suitable for battery-powered devices.

There is also provision for mesh nodes to access normal Bluetooth LE devices, so it's still totally possible to use Espruino devices as-is in a mesh. You just need to have fully-functional powered mesh nodes.

closed 2018-10-29 by espirate

gfwilliams, 2018-10-29:
I'm a bit confused - were you going to add support yourself? Or was this a request for it to be added for you?