espruino/Espruino #1545: setsockopt(SO_REUSPORT) failed on ESP32

Matthias-Hermsdorf, 2018-10-29:
Hi,

I want to have a Websocket Server on an ESP32.

    function onPageRequest(req, res) {
      res.writeHead(200, {'Content-Type': 'text/html'});
      res.end("Hello World");
    }

    var server = require('ws').createServer(onPageRequest);
   server.listen(80);

Before 1.99 I had to hardware reset the board before every upload. The socket doesn't got released.
Since 1.99 the socket got reused and I could send new Versions to the board without hardware reset. But after save() my client got no answer any more.
Now I have 2v00.27 and at every send I get an setsockopt(SO_REUSPORT) failed message.

The http server is working fine. The socket server is not.

Matthias-Hermsdorf, 2018-10-29:
This binary doesn't show the warning and sending code works.
But save() saves the code and unfortunatly afterwards it works no more.

wilberforce, 2018-10-29:
@opichals any ideas?

Matthias-Hermsdorf, 2018-10-31:
My fault. 

after reading again http://www.espruino.com/Saving I writen the server initialisation inside a init function, which runs once after copy and then on E.on('init', init); End everything works as wanted.

closed 2018-10-31 by Matthias-Hermsdorf