espruino/Espruino #1572: console history lags when setInterval()is set

fanoush, 2018-11-25:
When I use setInterval(), serial console command history starts to be very slow, when pressing up/down key I need to wait sometimes for ~2 seconds. it does not matter how long the interval or how complex the code is. even `setInterval(function(){},3000,true)` executed right after reset() slows it down.

Editing current line works fine, there is no lag, just up/down.

This is on nRF52.
```
>reset()
=undefined

 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v00.94 (c) 2018 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>1
=1
>2
=2
>3
=3
>4
=4
>5  
=5
>setInterval(function(){},3000,true) 
=1
>clearInterval(1)
=undefined
>
```
when I press up after setInterval it lags for 2 seconds before it prints something. Keys are batched so it skips few lines up if I press more keys, same down. After clearInterval(1) it is ok again.

gfwilliams, 2018-11-26:
Have you tried connecting from a different computer? And out of interest what board is this?

Just tried it here and it works fine. As you say, an interval like that shouldn't have any noticeable effect on performance.

What you may be hitting is Espruino switching to a low power BLE mode (~200ms intervals) after ~1 minute of inactivity. Pressing a few keys in quick succession should bring it back to the fast 7.5ms though

fanoush, 2018-11-26:
This is serial terminal with usb to serial adapter, speed is 38400. It is my build for the nRF52832 based DS-D6 watch - the "Espruino on your watch!" forum thread.

So if you cannot replicate it with other nRF52832 based device with serial console, then I guess it is something on my end. However there is nothing special about the build, board file is here https://github.com/fanoush/ds-d6/blob/master/espruino/boards/DSD6.py and there is no custom code  running yet when this already happens.

I'll try default  9600 speed. Also I'll try to tum bluetooth off completely via NRF.sleep() if it makes a difference. I didn't douch the deep sleep setting and the interval does not matter, even shorter one do this too. Anyway thanks for testing I'll try more.

gfwilliams, 2018-11-26:
Thanks - you could always just check out the RELEASE_2V00 tag and try that build.

Deep Sleep is ignored on the nRF52, so there's not much different you can do really. Since it's your own build and you were looking at adding JS code to run at the start, is it possible that you have added your own JavaScript code in addition to that `setInterval` that is actually taking up all the CPU time?

fanoush, 2018-11-26:
> is it possible that you have added your own JavaScript code in addition to that setInterval that is actually taking up all the CPU time

No, not yet, I even removed my modified modules with this build and load them later over console (as they are in progress anyway). Also I did not run save() yet so I guess nothing else is running. So this is basically just MDBT42Q build with few changed pins. Will investigate more and let you know.

fanoush, 2018-11-26:
Well, when testing it more it is s pretty consistent when done like described above and exactly one setInterval is scheduled however when more are scheduled it goes away, and when clearing it back to one being active it reappears. This is pretty strange.

gfwilliams, 2018-11-27:
Just tried here on an nRF52832DK with exactly the build you mention:

```
Serial1.setup(38400,{rx:D8,tx:D6})
// reconnect at 38400
setInterval(function(){},3000,true);
```

Everything works great for me. I think this could be some issue at your end, especially given the extremely specific point at which it happens.

fanoush, 2018-11-27:
>  I think this could be some issue at your end, especially given the extremely specific point at which it happens.

Can be, however there is nothing specific to my setup. Two different boards - one being just the bare nRF52832 breakout board and nothing else, even the official build, only the code above entered, two different usb to serial adapters.

OK, I'll try different computer, different terminal emulation maybe even diferent power source. However the issue starts with first setInterval() and stops with matching clearInterval() so this is internal state change  to espruino, I am not sure how the connected client could behave differently related to this.

> Just tried here on an nRF52832DK

Can you paste complete terminal session what you did from poweron of the board?

gfwilliams, 2018-11-27:
I'm not doing it again, but what I posted up is what I did.

setInterval, then used up/down and it was instant, as was hitting 'enter' a few times

gfwilliams, 2018-11-28:
I used exactly the build you suggested, from the link you posted. The DK board shouldn't be any different to yours - it just has the UART on the same pins as the MDBT42Q build which makes it easy - so the only difference would be the pins that are used.

> because uart rx does not have buffer for 3 characters

It has a buffer for around 1000, so that's unlikely to be the issue :)

You could try http://www.espruino.com/Reference#l__global_setSleepIndicator - it might give you some idea whether Espruino is waking up to service the characters coming in.

fanoush, 2018-11-28:
Meanwhile I figured out the 2 character buffer here https://github.com/espruino/Espruino/blob/master/targets/nrf5x/jshardware.c#L106  is in fact used as two single character buffers here https://github.com/espruino/Espruino/blob/master/targets/nrf5x/jshardware.c#L979 so it would be harder to increase that easily. Looks like there is a lot of event handling (and context switches?) to receive every single character.

> It has a buffer for around 1000, so that's unlikely to be the issue :)

Interesting, where it is defined? Apart from those two single character buffers you initialize there should be only hardware FIFO mentioned here https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.0.0/group__nrf__drv__uart.html?cp=4_0_4_6_8_26_2_16#gad56c8170e598afdddfc281bc13f3e8df ? Otherwise it would be wasting precious memory.  And btw with 1000 characters I probably wouldn't have issues with pasting few lines via clipboard. Even very small fragments are not received correctly from clipboard and are missing parts so I am sending files via ascii-xfr -c 1. Anyway it is indeed not because of the buffer.

As for the sleep indicator I'll try to add LED or something but sleep looks quite probable. Also forgot to say that the interval does relate to the slowdown. If I schedule setInterval(()=>{},500) instead of 3000 the up/down is faster and the delay seems to be 0.5s and with e.g. 200 it is even better.

gfwilliams, 2018-11-28:
In `jshSleep` in `jshardware.c` try moving `hadEvent = false;` to **after** the while loop and see if that helps.

The 2 character buffer is *extremely* internal, and very unlikely to be the issue. Characters come out via IRQ and go into a buffer via jshPushIOEvent. The 1kb buffer is tagged and shared between all devices including BLE - hence not wasting memory.

> have issues with pasting few lines via clipboard. 

Are you sure it's not just that you have an actual hardware problem connecting the UART, like not having GND connected? I have zero issues pasting large amounts of code from the clipboard, so something is very broken for you.

Unless I can actually find this happening with a board that I am supporting, I can't help you more.

I can't afford to spend hours explaining every internal bit of Espruino to everyone who wants to port to their own devices and has problems.

fanoush, 2018-11-28:
> Are you sure it's not just that you have an actual hardware problem connecting the UART, like not having GND connected? 

I have both GND and 3.3V, and then RX,TX all from CP2104 usb to serial adapter and sometimes using black magic probe with also SWD IO and CLK.. In both cases RXD,TXD,GND is from same source. The issue with no flow control is quite well known and almost every terminal program has settings and even some default delay for this (well except picocom :-) but [there is a patch](https://github.com/npat-efault/picocom/issues/96) ) so you may even have this delay enabled without knowing it?  Also that's why you have the "Throttle Send" option in web IDE? Anyway, I did not test it so much what exactly was the issue, after first couple of  fails with various sizes I started using ascii-xfr which is even easier than selecting blocks with mouse. I only know all characters that were received were ok, there were just random parts missing. So no issue with missing bits or garbled characters. Just when looking at the command history  with cursor up, something was missing so I got syntax errors. But forget it, I'll double check, In fact I sometimes I get the bug/feature with 0~ and 1~  around pasted text described here 
https://unix.stackexchange.com/questions/196098/copy-paste-in-xfce4-terminal-adds-0-and-1
 so that may be also the case when I tried copy/paste.

> Unless I can actually find this happening with a board that I am supporting, I can't help you more.

OK, no problem. I don't need help as such. I was reporting a bug. You may of course freely ignore it, which may be understandable as it is quite obscure one. I already found plenty of workarounds so I don't really need this fixed.  I can imagine you have quite a lot of other work to do, so definitely don't feel obliged to help me or answer.

Anyway, will try the jshSleep suggestion, thanks.

fanoush, 2018-11-28:
Just to be sure I now tried FTDI  FT232RL usb to serial adapter and with this one it happens too in exactly the same way.  I also tried copy/paste, made sure to not have that bracketed paste mode enabled and pasting few lines works fine. 

I have also added LED and used setSleepIndicator(pin) and it works nice as indicator and is mostly off unless typing. setInterval also blinks a led very briefly when it fires. However this issue reported goes away after running setSleepIndicator(D26) and it returns back after  setSleepIndicator(undefined). So this is enough to change the timing so it does not happen.

Will rebuild with the suggested jshSleep change.

gfwilliams, 2018-11-29:
> perhaps only internal to this one file and is just breakSleepLoop marker

Yes, it is. Otherwise every Bluetooth LE event would cause the interpreter to wake - when we only care about some of them.

Also when data comes in, the device wakes, it's handled by IRQ immediately and shoved into the big queue - so not an issue there.

I'll stick those changes in later as I want to do some tests before/after. I don't want this to end up causing an extra wake cycle the majority of the time.

fanoush, 2018-11-29:
> I'll stick those changes in later as I want to do some tests before/after. I don't want this to end up causing an extra wake cycle the majority of the time.

OK. However by clearing it right before starting the wait loop you are basically ignoring any new events/interrupts that happened since last wakeup. Setting it right after breaking the loop is safest way to not to miss anything. If any nrf event/interrupt happens after resuming from jshSleep you should not go to sleep next time you enter it because it may still not be handled.

So currently the hadEvent is used as "did any real event hapened **during sleep** so I should resume?"  However another issue is "are all previous events/interrupts already handled so I can safely enter the sleep?". Moving hadEvent = false; after exiting the loop helps to answer also this question however maybe indeed for the price of another empty loop. But this should be quick and is right thing to do - one should not enter sleep with pending events.

Other solution might be to have each event marked as handled. Then before going to sleep you could check if all events that happened since last wakeup are already handled. I guess you don't have this yet.

fanoush, 2019-01-14:
> I have added
> 
> ```
> if (ch == 14) { // Ctrl-n=down
>   inputState=IS_HAD_27_91;ch=66;
> }else if (ch == 16) { // Ctrl-p=up
>   inputState=IS_HAD_27_91;ch=65;
> }
> ```

Not sure what happens with this bug - I have the move of `hadEvent=false;` after the loop in my build and it works perfectly and it fixed this issue, however I had to remove the ctrl+p/n code quoted above since it somehow affected Espruino web IDE. When I connected to the device the IDE always executed one previously entered line as part if its connection initialization and sometimes failed to recognize the device (depending on contents of previous line). After removing this code it works again! So perhaps the 16 character code is already used for something and/or web IDE sends it to console for some reason? So while it seemed like good idea initially  to add those typical cursor movement Ctrl keys it looks like it can break something. And as the cursor keys are working fine now it is no longer needed. So this is just a warning about the web IDE vs ctrl-p (code 16=0x10) behaviour. When googling it now it may be perhaps related to [this meaning of the code](https://en.wikipedia.org/wiki/C0_and_C1_control_codes#DLE)?

gfwilliams, 2019-01-14:
Char code 16 turns echo off for the current line in Espruino if it's issued right at the start of the line, so that's probably what you're hitting?

commit 7d3b1efa (2019-03-19): nRF5x: Don't sleep if we received any events since last sleep (fix #1572)
closed 2019-03-19 by gfwilliams