espruino/Espruino #1596: auto-add Connection:close header

gfwilliams, 2019-01-11:
http://forum.espruino.com/conversations/329528

Looks like since reporting HTTP/1.1 when as a server, the default of most clients is to leave the connection open, which isn't what we want (usually).

gfwilliams, 2019-01-21:
Actually having trouble reproducing this issue on Chrome/Firefox, but `curl` definitely seems unhappy without a `close`.

```
page = `<!DOCTYPE html>
  <head><link rel="icon" href="data:,"></head>
<body>Hello world!</body></html>`;
var http = require("http");
function handle(req, res) {
  print("Opened");
  res.on('close',_=>print("Closed"));
  res.writeHead(200);
  res.end(page);
}

http.createServer(handle).listen(8080);
```

closed 2019-01-21 by gfwilliams

MaBecker, 2019-01-21:
On ESP8266 it cause blocking sockets, for sure.

gfwilliams, 2019-01-21:
I don't get why Chrome doesn't leave the socket open when I'm running on localhost with a Linux build though. Maybe there's something more to it, like it's got to be website that's actually doing something

MaBecker, 2019-04-03:
adding "Connection:close" allowed me to hammer Espruino on ESP8266 with curl to reuest some JSON without any errors.

```
0001 20:33:11 : {"ssid":"mySSID","pass":"myPass","broker":"myBroker","topic":"myTopic"}
......
1000 20:33:57 : {"ssid":"mySSID","pass":"myPass","broker":"myBroker","topic":"myTopic"}
```

Amazing: 1000 calls in 45 sec without errors with this esp8266 usb device.

[image]

@gfwilliams Do you have plans to sell Espruino sticker?

gfwilliams, 2019-04-04:
That's a neat looking little device...

> Do you have plans to sell Espruino sticker?

Eventually! I've been hoping to get a slightly more professional logo done first, then I'd like to be able to offer a few different branded things :)

MaBecker, 2019-04-04:
Like to have a cup with Espruino Logo and device on it.

commit 4e51cede (2019-01-21): Added HttpServerResponse.setHeader
            HttpServerResponse now automatically sends `Connection:close` unless overridden (fix #1596)