espruino/Espruino #1597: Q: WASM support?

grahamehorner, 2019-01-12:
Great work guys and gals, I’m wondering if there are any plans to support the WASM?
IMHO this give better size/performance when running on the target device as a WASM compiler would perform ahead-of-time optimisations and minifications, it could also open up additional language support given a number of language compilers support WASM as a target, ie. C# can be compiled to target wasm using the mono compiler, or why not run the .net core compiled code (without modification) using the mono runtime that is targeted for wasm ?

gfwilliams, 2019-01-14:
I don't think it's something that will ever go into core Espruino itself, but I guess it might go in the toolchain. Basically to get WASM you're going to have to have a toolchain set up on your PC in the first place, so why not do all the work on the PC and send the native code over - which is perfectly doable right now.

WASM sounds cool, but you've been able to compile quite a large subset of asm.js code into native code in Espruino for ages with http://www.espruino.com/Compilation and as far as I know nobody has used it at all - so I don't see how WASM (which is arguably even less approachable) will be any different. It's also not clear how it would integrate nicely with something like the Web IDE.

It's worth noting that there are plenty of options to get native code in Espruino already:

* http://www.espruino.com/Compilation will compile a subset of JS to very fast *optimised* native code for the ARM microcontroller
* You can write C code in-line with http://www.espruino.com/InlineC which will also produce optimised native code for the ARM microcontroller
* You can write ARM Thumb assembly directlt with http://www.espruino.com/Assembler
* You can upload assembled ARM Thumb code directly and use http://www.espruino.com/Reference#l_E_nativeCall to reference it - which is what the above 3 options do under the hood.

But yeah, if someone wants to jump in and write some an extension for https://github.com/espruino/EspruinoTools that will transform WASM in to ARM Thumb assembly then that'd be amazing.

closed 2019-01-14 by gfwilliams

grahamehorner, 2019-01-14:
IMHO to rule out support of wasm is wrong, firstly wasm not only saves on overall space, but can/is optimised ahead of time using advanced compiler techniques on/by larger pc compilers such as mono, this can and offten does result in better at runtime performance over asm.js source-to-source, secondly WebAssembly is in its binary form wasm is intended to work cross platform and/or in browser without the need for a host to have a JavaScript virtual machine, there is nothing stopping JavaScript accessing WebAssembly and/or code created as a WASM module or even using the WebAssembly textual format WAT 

FYI asm.js has been deprecated of favor of wasm and a number of #iot projects are now focused on providing WebAssembly support as well as JavaScript together with other high level support via WebAssembly see https://webassembly.org. 

https://webassembly.org/roadmap

I’d be happy to fork and contribute to supporting WebAssembly as I truly believe IMHO it can/will brings a lot of benefits to IoT space above and beyond just the size, performance improvements that can be achieved using a developer tool chain or deployment of pre built wasm modules that are acceptable via JavaScript

gfwilliams, 2019-01-14:
I'm not sure you read what I just wrote, so I'll try and be more clear:

It looks from your GitHub like you're doing a lot of work with ESP8266.

Espruino for some ESP8266 parts is a tight squeeze at the moment. To add WebAssembly support into the Espruino interpreter would mean removing functionality that many people use, in order to add something that:

a) very few people would use
b) would be implemented more easily, quickly and reliably if it was done as part of the EspruinoTools toolchain.

If someone (like you?) wants to implement it as part of the EspruinoTools toolchain so that it uploads binary code direct to Espruino then that would be great though and it'd magically become part of the IDE and CLI tools.

Just to make it even clearer, there's a wishlist here with WebAssembly on it (which you'd have found if you bothered searching): https://github.com/espruino/Espruino/issues/1302

You'll note that there's not a single +1 for it.

grahamehorner, 2019-01-14:
@gfwilliams while I am looking at the esp8266 this is not the only iot device I’m interested in or looking at, so while this true it’s not the complete picture!

I also find your reply so what harsh, and directed in anger towards myself; as such I won’t BOTHER YOU ANYMORE and I’ll be working with others that take a more pro active open and friendly approach to suggestions and/ideas and have a more open source community mindset.

I wish you luck with whatever you decide to do.