espruino/Espruino #1599: ESP32 having connection issues (espruino-2v01_esp32)

jgtaylor, 2019-01-15:
So, build release 2v00 connects and works fine.

build release espruino-2v01_esp32 has some issues. After reading through some PRs, and recent commits, I'm guessing it has more to do with moving to the Espressif 3.1 SDK, than anything else.

I thought I would check out the state of esp32's getHostByName, setHostname, etc. All of them work quite nicely. (thank you, @wilberforce !, btw) 👍 

What I discovered was the radio seems to drop out. Upon poking around, I noticed the output of `wifi.getStatus()` had changed, specifically, `wifi.getStatus().powersave`. it went from `none` to `modem`.

The behavior is that the wifi simply drops out, though `wifi.getIP()` shows an ip, `wifi.getStatus().station` shows `connected`, etc. As far as what's displayed, everything should be on. Instead, it outputs errors as below. (also, I have a `wifi.on("disconnected", ...` that output the reasons.

```
E (173464) event: post event to user fail!
E (176306) event: post event to user fail!
E (179146) event: post event to user fail!
E (181987) event: post event to user fail!
{
  "ssid": "X11",
  "mac": "00:00:00:00:00:00",
  "reason": "201"
 }
```

I don't mean to make this confusing, but the output from getIP(), etc., isn't always the same. Here's some output from a quick test just now:
```
>wifi.getIP();
={
  ip: "0.0.0.0",
  netmask: "0.0.0.0",
  gw: "0.0.0.0",
  mac: "30:ae:a4:12:3f:34"
 }
>wifi.getStatus();
={
  station: "NO_AP_FOUND",
  mode: "APSTA",
  powersave: "modem"
 }
>wifi.getDetails();
={
  status: "NO_AP_FOUND"
 }
> 
```
In my tests, only an ESP32.reboot(), or power off/on resets things to work again.

this may already be fixed upstream, but I thought I'd document it for posterity.

wilberforce, 2019-01-15:
> So, build release 2v00 connects and works fine.
Other people reported that this did not work.

I have seen the behaviour you describe - the Station sends disconnects and it does not reconnect  properly.

I wonder if resetting the modem power save mode in the connect might work?

I'm travelling at the moment so are not in a position to try this.  @jumjum123 do you have time?

jumjum123, 2019-01-16:
Don't have much time. Haven't had the problem, but will check extensive in my environment.
To give at least some first (OT ?)feedback
Looks like there are some problems in firmware around BT and WIFI
In Version 3.1 some BT functions in firmware are replaced by components in ESP-IDF. This was one reason, why size of binary growed that much with 3.1 Another reason was around Mesh, independent of usage in application.
Looking further to V3.2 in github, there are some new components around http etc. May be, these are some more functions to replace firmware.
Last not least, some entrys in ESP32 forum are pointing to memory leaks and other problems around WIFI 
From my point of view, looks like BT and WIFI are heavy restructured by Espressif. Information from Espressif about that is, let me say, challenging

jumjum123, 2019-01-16:
Test is running like this, no problems yet.
@jgtaylor from milky way ;-) do you have some source to reproduce the problem ?
```
var ssid = "JUMWare";var password = "my password";
var wifi = require("Wifi");
function connect(){
  wifi.connect(ssid, {password: password}, function(ap) {
    console.log("connected:", ap); 
  });
}
var i = 0;
connect();
setInterval(function(){
  if((i % 10) === 0) console.log(i,wifi.getStatus());
  i++;
},5000);
```
Running for 20 minutes now, and still I get
```
 {
  "station": "connected",
  "ssid": "JUMWare",
  "bssid": "34:81:c4:c0:d6:21",
  "channel": 6, "rssi": -41,
  "htMode": "HT40+",
  "authMode": "wpa2",
  "mode": "APSTA",
  "powersave": "modem"
 }
```

jgtaylor, 2019-01-17:
I do have some code that does this. It's a bit convoluted because it includes uploading some html, css & stuff for a wifi initial setup. I use Postman and curl to upload the files, and i've hard coded them all in. I haven't reduced the problem to anything more than running an http instance and connecting. Anyway, here's the code:
```javascript
var wifi = require("Wifi"), fs = require("fs"), http = require("http"), storage = require("Storage");

function dataStore(req, name) {
  var lastData = 0;
  var payload = "";
  req.on("data", function(data) {
    payload += data;
    lastData = lastData + data.length;
  });
  req.on("close", function() {
    if ( payload.length == parseInt(req.headers["content-length"]) ) {
      storage.write(name, payload);
    }
  });
}

function setupSystem(config) {
  if ( storage.list().filter((e)=>{return e == "ptCfg";}) ) {
    var _config = storage.readJSON("ptCfg");
  }
  if ( config.ssid !== "" ) { 
    if (config.ssid.length <= 32) { // max 32byte name length
      console.log('wifi.connect(config.ssid, {password: config.password})');
    }
  }
//  {
//    "ssid": "",
//    "wifiPassword": "",
//    "serverAddress": "",
//    "serverPort": "",
//    "serverPath": ""
//   }
}

function reqGet(req, res) {
  var q = url.parse(req.url, true);
  var type;
  var name;
  if ( q.pathname == "/index.html" || q.pathname == "/") {
    type = {'Content-Type': 'text/html'};
    name = "index";
  }
  if (q.pathname == "/blackcat.svg") {
    type = {'Content-Type': 'image/svg+xml'};
    name = "blackcat";
  }
  if (q.pathname == "/paper-tiger.css") {
    type = {'Content-Type': 'text/css'};
    name = "css";
  }
  res.writeHead(200, type);
  res.write(storage.read(name));
  res.end();
  if (q.query) {
    setupSystem(q.query);
  }
}

function reqPost(req, res) {
  var fname = req.url.replace("/","");
      if (fname == "index.html") {
        dataStore(req,"index");
      }
      if (fname == "favicon.ico") {
        dataStore(req, "favicon");
      }
      if (fname == "blackcat.svg") {
        dataStore(req, "blackcat");
      }
      if (fname == "paper-tiger.css") {
        dataStore(req, "css");
      }
    res.writeHead(200, {"Content-Type": "text/html"});
    res.write("it seems to have worked.<br>");
    res.end();
}

function onPageRequest(req, res) {
  if (req.method == "POST") {
    reqPost(req, res);
  }
  if (req.method == "GET") {
    reqGet(req, res);
  }
}

wifi.on("connected", () => {
  console.log("wifi connected");
  var h = http.createServer(onPageRequest).listen(80);
});
wifi.on("disconnected", (detail) => {
  console.log(detail);
});
if (wifi.getStatus().station == "connected") { 
  h = http.createServer(onPageRequest).listen(80); 
} else {
  wifi.connect("XXXXX", {password: "XXXXXX99"});
}
```

this code produced the following output:
```
>
:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
Found ESP32, 2v01
>
Connected to /dev/cu.SLAB_USBtoUART
No errors. Minifying 2416 bytes to 1764 bytes
>WARNING: Scan stop failed
WARNING: set rssi scan not implemeted yet
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v01 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
>E (55173) wifi: ieee80211_output.c 853
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2588
load:0x40078000,len:7192
load:0x40080000,len:5288
entry 0x40080278
E (29) boot: ota data partition invalid, falling back to factory
E (517) spiram: SPI RAM enabled but initialization failed. Bailing out.
E (555) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v01 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
E (1070) phy_init: store_cal_data_to_nvs_handle: store calibration data failed(0x1105)
WARNING: Wifi:startMDNS - espressif
No errors. Minifying 2431 bytes to 1781 bytes
>WARNING: Scan stop failed
WARNING: set rssi scan not implemeted yet
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v01 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
>E (187873) event: post event to user fail!
E (190288) event: post event to user fail!
E (192703) event: post event to user fail!
E (195119) event: post event to user fail!
{
  "ssid": "X11",
  "mac": "00:00:00:00:00:00",
  "reason": "201"
 }
>wifi.getAPStatus()
Uncaught Error: Function "getAPStatus" not found!
 at line 1 col 6
wifi.getAPStatus()
     ^
>wifi.get
getAPDetails        getAPIP
getDetails          getHostByName
getHostname         getIP
getStatus
>wifi.getStatus();
={
  station: "NO_AP_FOUND",
  mode: "STA",
  powersave: "modem"
 }
>wifi.getDetails();
={
  status: "NO_AP_FOUND"
 }
>wifi.getIP()
={
  ip: "0.0.0.0",
  netmask: "0.0.0.0",
  gw: "0.0.0.0",
  mac: "30:ae:a4:12:3f:34"
 }
> 
```

jgtaylor, 2019-01-19:
The version I flashed was the one on the "Get Espruino" download page (https://www.espruino.com/binaries/espruino_2v01_esp32/). I don't have access at the moment, but as soon as I do, I'll get a `process.env` output.

I should also attempt to narrow down a re-creation of the issue. If I can, I'll update here.

Re, distance to wifi - i'm about 4.5 meters away from the AP, with a plaster wall in between.

barbiani, 2019-01-22:
I noticed that trying to connect to ESP_203D1D makes espruino fail.

Happened as soon as I try to connect:
>abort() was called at PC 0x40084b1f on core 0

or the normal error we see:
E (36210) event: post event to user fail!
E (39052) event: post event to user fail!
E (41894) event: post event to user fail!
E (44737) event: post event to user fail!

wilberforce, 2019-01-29:
Looks like this fix might address the issue:

Fix issue which causes an ESP32 in station mode to disconnect from certain APs which announce two SSIDs

https://github.com/espressif/esp-idf/releases/tag/v3.1.2

When I get a chance I'll update espruinobuildtools to  try 3.1.2

@jgtaylor - what brand of router are you using? Does it broadcast 2 ssids ?