espruino/Espruino #1600: ESP32 flashfs issues (espruino-2v01_esp32)

barbiani, 2019-01-17:
I am seeing fs corruption after a few file writes and unlinks that re-formatting/power cycle is not able to recover and I have to reflash the firmware.

Overwriting a file once also corrupted the filesystem.

Version 1.99 is good.

gfwilliams, 2019-01-21:
Do you have any example code to reproduce, or links for a forum post where this is discussed?

barbiani, 2019-01-22:
v2.01 build from the website failed when calling file.write() on the same file twice in a row.

Build 2.01.8 is not.

E.openFile of my http file send is returning undefined after the browser is refreshed a few times. (Since 1.99 that I tested).

I am seeing things like:
```
>var h = E.openFile("foo.txt", "r");
=File: {  }
>h.close()
=undefined
>h = E.openFile("foo.txt", "r");
=undefined
>process.memory()
={ free: 1532, usage: 668, total: 2200, history: 17,
  gc: 0, gctime: 2.329 }
>reset()
=undefined
WARNING: Scan stop failed
WARNING: set rssi scan not implemeted yet
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v01.8 (c) 2018 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>var h = E.openFile("foo.txt", "r");
=File: {  }
> 
```
So after a few refreshes the http server can not serve files anymore.

wilberforce, 2019-01-29:
@barbiani

Your example does File reads. Do you have a simple reproducible example that shows the file writes failing?

gfwilliams, 2019-01-29:
What if you read the file *after* calling `process.memory()`? This will reallocate the free list which may help with loading the file.

wilberforce, 2019-01-29:
Please see this issue:

https://github.com/espruino/Espruino/issues/1559

And in the forum note - see comment 11 - it is a work around that solved an issue in the past

MaBecker, 2020-03-13:
@barbiani is this still a issue with the latest release?

GaryOtt, 2021-03-12:
This issue looks like a duplicate (or more specifically, the original) of https://github.com/espruino/Espruino/issues/1768 which was closed.

gfwilliams, 2021-03-15:
Thanks! Yes, it does look similar.

closed 2021-03-15 by gfwilliams