espruino/Espruino #1605: Slower Bluetooth connect on Espruino 2v01?

gfwilliams, 2019-01-24:
Mentioned at http://forum.espruino.com/conversations/330001/#comment14589122

Worth looking at. Does `DYNAMIC_INTERVAL_ADJUSTMENT` end up setting the connection interval back to low speed if there's no activity because Bluetooth isn't connected? Maybe - if so we should ensure that the connection speed goes back to `BLE_DYNAMIC_INTERVAL_HIGH_RATE` on disconnect and stays there.

olliephillips, 2019-01-24:
Possibly a false alarm. Please see http://forum.espruino.com/conversations/330001/#comment14589340

gfwilliams, 2019-01-24:
No problem - it's actually still worth me looking into as I think it is an issue. It may not be making it *much* slower on the whole, but it might be taking ~1 sec sometimes when it could be doing it in 0.2 sec or so.

commit 329d026a (2019-02-19): nRF52: Fix slow Bluetooth connection if previously disconnected while using low power connection interval (fix #1605)
closed 2019-02-19 by gfwilliams