espruino/Espruino #1612: ESP32: setSNTP implementation and setInterval

wilberforce, 2019-02-02:
Discussed in forum here:

http://forum.espruino.com/conversations/330214/#comment14602067

The esp-idf updates the system time - however `jswrap_interactive_setTime` needs to be called to update the last `jsiLastIdleTime`.

unfortunately the api does not provide a callback - in the ESP8266 implementation a system timeout is added to poll every 100ms:

https://github.com/espruino/Espruino/blob/5bc873ae381b39504e264d73576535e40b9c0104/libs/network/esp8266/jswrap_esp8266_network.c#L1140-L1184

MaBecker, 2020-03-13:
@wilberforce how should this be implemented?

jumjum123, 2020-03-17:
FYIO, looks like sntp got a totally new structure in V4.
Compared header from V3.1.3 with V4.0 and its totally different.
V3.1.3 was from Author: Frédéric Bernon, Simon Goldschmidt and V4.0 is
Espressif

MaBecker, 2020-08-30:
Hmm @jumjum123 it is working on my site, running on 2v06.xx

```JS
Date()
//=Date: Thu Jan 1 1970 00:00:07 GMT+0000
E.setTimeZone(2)
//=undefined
Date()
// =Date: Thu Jan 1 1970 02:00:23 GMT+0200
require('Wifi').setSNTP('0.de.pool.ntp.org','2');
//WARNING: SNTP: 0.de.pool.ntp.org 2
//undefined
Date()
// =Date: Sun Aug 30 2020 13:32:19 GMT+0200
```

Should this be closed?

closed 2022-11-22 by MaBecker