espruino/Espruino #1615: Graphics.clear should reset font/align/etc?

gfwilliams, 2019-02-12:
It's always a bit frustrating if you're doing Graphics and you set to a large font - then next time around the code that used to draw a small font now draws it large.

I think maybe `.clear()` should reset the font options? But probably not color or rotation.

Any thoughts?

gfwilliams, 2019-02-19:
@IrregularShed @MaBecker do you have any feelings about this either way?

IrregularShed, 2019-02-19:
I'm not sure. There's arguments both ways, depending on how the graphics are being used - if I'm creating a display with one value being reported it's handier to not have to think about the font size each iteration, whereas if I were to have one large value and two smaller ones (for example) I can see how the automatic reset would help.

Any mileage in having an optional parameter passing through flags? That might be a little excessive, I guess.

gfwilliams, 2019-02-19:
I think an optional parameter might be overkill, but maybe `g.reset()` or something that would set all these items to their default values would be better? Doing `g.clear();g.reset()` isn't the end of the world, but it's a lot better than having to reset fonts/etc manually each time.

IrregularShed, 2019-02-19:
Seems reasonable!

wilberforce, 2019-02-19:
remember you could chain...   `g.clear().reset()`

How about  `g.clear(1)`

gfwilliams, 2019-02-20:
@wilberforce thanks - `g.clear(1)` sounds like a perfect solution

MaBecker, 2019-02-20:
g.clear(1) - perfect for me too.

commit d0fa7167 (2019-02-21): Allow `Graphics.clear(true)` to reset state (font, color, etc) to default as well as clearing the screen (fix #1615)
closed 2019-02-21 by gfwilliams