espruino/Espruino #1619: bugs found by our bug scanner

lqiulin, 2019-02-19:
Hi,

Our bug scanner has reported  something that could be improved.
Bug triggering files are attached.

**Bug-1:div/mod-zero**

1. in function `jsserialSoftwareFunc`,[jsserial.c#L50](https://github.com/espruino/Espruino/blob/master/src/jsserial.c#L50)

```c++
JsSysTime bitTime = jshGetTimeFromMilliseconds(1000.0 / inf->baudRate);
```
Divisor: inf->baudRate
Result: Could be 0, Please Check.

2.in function `wrapAround`,[jsutils.c#L671](https://github.com/espruino/Espruino/blob/master/src/jsutils.c#L671)
```c++
val = val / size;
```
Divisor: size
Result: Could be 0, Please Check.

3.in function `i2c_initstruct`,[jsi2c.c#L142](https://github.com/espruino/Espruino/blob/master/src/jsi2c.c#L142)
```c++
inf->delay = 250000/i->bitrate;
```
Divisor: i->bitrate
Result: Could be 0, Please Check.

**Bug-2:Mem_op_argument**

1.in function `mbedtls_rsa_rsassa_pkcs1_v15_sign`,[rsa.c#L1103](https://github.com/espruino/Espruino/blob/master/libs/crypto/mbedtls/library/rsa.c#L1103)

```c++
memcpy( sig, sig_try, ctx->len );
```

[memcpy] is a memory operation function using tainted data: [ctx->len]

2.in function `jsvNewArrayBufferWithData`,[jsvar.c#L4006](https://github.com/espruino/Espruino/blob/master/src/jsvar.c#L4006)

```c++
memcpy(dst, data, (size_t)length);
```
[memcpy] is a memory operation function using tainted data: [length]

3.in function `jsvMalloc`,[jsvar.c#L4028](https://github.com/espruino/Espruino/blob/master/src/jsvar.c#L4028)

```c++
memset(p,0,size);
```
[memset] is a memory operation function using tainted data: [size]

gfwilliams, 2019-02-19:
Thanks for the report! Please could you provide a little more info about what you mean by `is a memory operation function using tainted data:`? As far as I can see the variable given (length/size/etc) is passed in as an argument and isn't changed?

I just checked the serial baud rate and it can never be 0, but it's possible that I2C bitrate isn't range checked so I could add something for that.

stuartly, 2019-02-19:
@gfwilliams Thanks for your response! 

We developed a taint analysis based static analysis tool to identify security-sensitive operations such as divide-zero, mod-zero, array-index-access, and sensitive function call related bugs without proper checks its arguments.

Memory operation functions are defined as the memory-related functions such as memcpy, memmove, etc and string-related function such as strcpy, strcpy_s, etc. If its argument could be affected or changed by outside input, we define it as tainted data.

Thanks
Lingyun Situ

gfwilliams, 2019-02-19:
Ok, thanks - I think in the cases I've looked at there's nowhere where the functions are called with the wrong values, however I have now added some assertions to trap any of those issues on debug builds.

I've also stuck in some checks for SPI/I2C initialisation that could have had invalid baud rates specified.

closed 2019-02-19 by gfwilliams