espruino/Espruino #1622: additional CFLAGS shrinking binary by 1900 bytes

MaBecker, 2019-02-24:
Reading espressif/ESP8266_NONOS_SDK issues: https://github.com/espressif/ESP8266_NONOS_SDK/issues/209#issuecomment-466655556

without
```
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user2.elf
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  3 .irom0.text   000755a4  40201010  40201010  00009e90  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00007034  40100000  40100000  00002e58  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf >espruino_esp8266_user1.lst
  8 -rwxr-xr-x  1 markbecker  staff    2220 24 Feb 11:14 eagle.app.v6.data.bin
944 -rwxr-xr-x  1 markbecker  staff  480676 24 Feb 11:14 eagle.app.v6.irom0text.bin
 24 -rwxr-xr-x  1 markbecker  staff    9416 24 Feb 11:14 eagle.app.v6.rodata.bin
 64 -rwxr-xr-x  1 markbecker  staff   28724 24 Feb 11:14 eagle.app.v6.text.bin
** user1.bin uses  521108 bytes of 831488 available
```

CFLAGS += -fno-cse-follow-jumps
CFLAGS += -fno-tree-fre
CFLAGS += -fno-ipa-sra
CFLAGS += -fno-signed-zeros -fno-trapping-math -fassociative-math
CFLAGS += -fno-caller-saves
CFLAGS += -fno-move-loop-invariants
CFLAGS += -fno-tree-tail-merge
CFLAGS += -fno-guess-branch-probability
CFLAGS += -fno-tree-copy-prop

after adding those flags:

```
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user2.elf
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  3 .irom0.text   00074e24  40201010  40201010  00009e90  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00007030  40100000  40100000  00002e58  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
To disassemble: xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf
xtensa-lx106-elf-objdump -d -l -x espruino_esp8266_user1.elf >espruino_esp8266_user1.lst
  8 -rwxr-xr-x  1 markbecker  staff    2220 24 Feb 11:14 eagle.app.v6.data.bin
936 -rwxr-xr-x  1 markbecker  staff  478756 24 Feb 11:14 eagle.app.v6.irom0text.bin
 24 -rwxr-xr-x  1 markbecker  staff    9416 24 Feb 11:14 eagle.app.v6.rodata.bin
 64 -rwxr-xr-x  1 markbecker  staff   28720 24 Feb 11:14 eagle.app.v6.text.bin
** user1.bin uses  519188 bytes of 831488 available
```

What do you think?

gfwilliams, 2019-02-25:
Great - I'd say go for it then.

MaBecker, 2019-03-03:
result for ESP8266_BOARD is similar, 1712

    user1.bin uses  460788 bytes of 479232 available

to 

    user1.bin uses  459076 bytes of 479232 available

wilberforce, 2019-03-03:
@MaBecker 
Have you tested the new binaries? Do any of these flags have an affect on alignment - as this could trip up here.

MaBecker, 2019-03-03:
Yes, tested on a ESP8266 board.

@wilberforce Any special stuff I should include in the tests?

wilberforce, 2019-03-04:
Nothing comes to mind

commit 31e0fecc (2019-03-07): Merge pull request #1627 from espruino/ESP8266_cflags

add CFLAGs to shrink build size for ESP8266 (fix #1622)
commit d812594c (2019-03-07): add CFLAGs to shrink build size for ESP8266 (fix #1622)
closed 2019-03-07 by gfwilliams